theyosh/TerrariumPI #494: [FEATURE]: Multiple light & heat channels

WorriedTitan, 2020-12-26:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
For Light and Heat, I would like to be able to create multiple "Channels" that operate independently.  A Channel can be visualised as a pairing of an input condition (either hard coded like time, or sensor driven like temperature) and an output (like a relay or a MOSFET dimmer).  

The reason I would like this is to more closely match the natural environment, for example, Light is a different colour (Kelvin) in dawn and dusk (2,500K) vs during the day (6,500K).  The separate channels should allow us to control Light Colour and Intensity by ramping up or down different light sources at different times.

**Light:**

Dawn and Dusk are typically 2,500 Kelvin light
Midday is typically 6,500 Kelvin light

To simulate Dawn / Dusk more accurately, you would need to have two channels to drive the "Light" elements of the vivarium.  You can get very bright 2,500K and 6,500k LED floodlights.  Both would need to be separately connected to AC Dimmer circuits

Assume Dawn is from 6am - 7am and dusk is from 6pm - 7pm.  The timing of the channels could work as follows:

6am: 2,500K channel ramps up from 0% to 100% at 7am (configurable ramp up)
7am: 6,500k Channel ramps up from 0% to 100% at 9am (configurable ramp)
4pm: 6,500K Channel ramps down from 100% to 0% at 6pm (configurable ramp)
6pm: 2,500k Channel ramps down from 100% to 0% at 7pm (configurable ramp)

Hardware wise the RPI can control a large array AC Dimmers so this should not be a limiting factor.  LEDs are highly efficient and so Temp can be dealt with separately.

**Heat:**

Desert species require extremely hot basking spots and the ability to escape the heat, so having a heat bulb and thermostat in one end isn't always the best solution.  This is especially true when dealing with larger enclosures.  There are typically three heat zones, Basking (42 degrees C), Ambient (28 degrees C), Cool (25 degrees C).  The other complexity with Reptile Heat is that they can see IR and UV light, so just using LEDs (no UVB, UVA or IR) is coloured badly for them and will cause health and eyesight issues over time.  Not many dimmable bulbs can provide a full spectrum, so various bulbs should be used.

Channel Basking: Dimming Thermostat that just controls the Basking site.  On the dimmer are typically a Halogen Bulb and a IR projector like an Arcadia Deep Heat Projector.  Both can be driven off the same dimmer / thermostat pair.  The Halogen Bulb provides UVA and some IRA, The Deep Heat Projector provides IRA, B & C

Channel Ambient: A temperature probe in the middle of the back wall of the vivarium measures ambient temperature.  It controls a UVA rich ambient heat source (like an incandescent flood light) on a separate dimming / thermostat pair.  This UVA light source will balance the UVB (timer switched) and Light (LED, see above) for reptile eyes as well as beefing up the temperature gradient in big tanks

Channel Cool: A temperature probe measures the cool end.  If temperatures get too high, this should be able to trigger a relay circuit that turns on venting fans for cooling

I have experimented in the past and these three zones can be managed independently.  The temptation is to ask for the ability to set a master channel and assign slave channels, which would be cool, but not necessary for the minimal viable product.

**Additional context**
I am building a 2m x 1m x 1.5m vivarium for a Bearded Dragon and want to simulate nature as closely as possible, which requires multiple "channels" for lighting and heating, similar to marine aquarium systems.  This enables the creation of temperature zones as well as the ability to vary the colour of light (in addition to the intensity of light)

AndrejWeber, 2020-12-26:
Nice Idea,  
but, 
the Raspi can only use two different PWM signals at hardware-side.
It is possible to do software-PWM on each pin, but this will increase the cpu usage to a unusable state, if you have several saftware-PWM signals with more then 100Hz.
I dont know, if there are I2C dimmer, that can handle this, or a riser card, to have more hardware-pwm pins.

I can imagine, that you can use just one PWM pin for all light. you have to build a logic of 2 relais, which will route the PWM signal at the right time to the right LED. The first relais will change the PWM signal between input for 2500k or 6500k.
the second relais will close the sircuit after PWM will reach 100% and let the 2500k LED just work for 100 %
So ramp up the 2500k LEDs from 6 to 7am, then the second relais will close the circuit from a 3.3V pin to input of dimmer hardware and let them work at 100%.
then you have to set the PWM to 0% and close the first relais to ramp again from 0 to 100% for 6500k LEDs.
at 4pm you have to ramp down the PWM, then open first relais, set PWM again to 100% and after opening relais two, again ramp down to 0% for 2500k LEDs.
with this logic you can dim more than just two independent LED colours, but with the limitation, that only the last ramping up colour could stay dimmed to another percentage than 100% .
I do not think, that terrariumPi can handle this logic, but you can write your own code, if you do not have to change the intensity, or just time to start ramp and so on.

now you can use the second PWM pin for heating. I think, you also can use one or two software-PWM additionally, depending on RasPi version

WorriedTitan, 2020-12-27:
Thank you so much for you long and detailed reply, I will go through today. I just wanted to reply quickly and say that breakout boards for PWM do exist for robot projects where multiple servos need to be used:

PCA9685 16 Channel 12 Bit PWM Servo Driver for Arduino and Raspberry Pi (MEHRWEG) - as an example. Each of the outputs can be directly connected to a MOSFET AC Dimmer to drive LED or Heat Lights.

ModestZombie, 2020-12-27:
Thought I would chime in since I've looked into this same idea. I have that same module, [PCA9685](https://www.adafruit.com/product/815), and I had it with the same hopes that I could find a way to get past the 2 PWM hardware limit that the Pi has. However, that module that you recommended [RobotDyn AC Dimmer](https://robotdyn.com/ac-light-dimmer-module-1-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.html) wouldn't be compatible. The Pi & PCA9685 are missing a 'hardware interrupt pin' of sorts. From the product page: '_Dimmer is connected to Arduino controllers via two digital pins. First (Zero) to control the passing of Phase Null of AC, which is used to initiate the interrupt signal. Second (DIM/PSM) to control (dim) current.'_ 

I imagine using the dimmer as found [here](https://www.amazon.com/dp/B06Y1DT1WP/ref=cm_sw_em_r_mt_dp_oHm6Fb3NQ0BQK) or [here](https://www.tindie.com/products/bugrovs2012/pwm-2a-ac-light-dimmer-module-50hz-60hz-120v-240v/) would be more suitable. Even though that dimmer supports a 10KHz frequency, I would imagine that the max supported PWM frequency would be that of the PCA6985, so 1.6KHz, Which shouldn't be a problem, using that same dimmer I'm able to get steady frequencies at about 300Hz for controlling a 65W BR30 flood light. 

So the setup would look like: rPi -> Connects to PCA6985 via I2C -> connects to Dimmer via PWM

But, I have no idea how this all works out on the coding side of things, and haven't explored idea this much since I figured I better get my snake's setup with just the 2 dimmers first THEN I'll come back to this idea lol

WorriedTitan, 2020-12-27:
Awesome thank you, I will look into those modules!

ModestZombie, 2021-01-01:
I had a bit of time to research this tonight: As it stands, this module is integrated with TerrariumPi but it hasn't really been properly tested, according to [Issue#331](https://github.com/theyosh/TerrariumPI/issues/331). The kicker about this module: **Each PWM runs completely independently but they must all have the same PWM frequency.** 

So what does this mean?
-The way I see it, you are going to want to use the same dimmer module for however many channels you are wanting to use. Otherwise, some modules have reversed polarity, and would require the coding to be different per channel, and don't know if it's feasible. 
-On top of that, this may produce mixed results, depending on what you are controlling. If you are controlling different types of light bulbs, it's possible some could flicker while others work great. I imagine if they are the same lightbulb or devices, this wouldn't be too much of an issue. 

If I'm understanding the background logic correctly, this is sending a PWM signal at 1KHz. I have a couple of [these dimmers](https://www.tindie.com/products/bugrovs2012/pwm-2a-ac-light-dimmer-module-50hz-60hz-120v-240v/) handy, so I'll have to pick up where [issue#331 left off](https://github.com/theyosh/TerrariumPI/issues/331#issuecomment-579348152) - with some good old fashioned testing (and hopefully not blow another dimmer…. Or the Pi lol). It may take me up to a week but I'll post my findings. And someone please correct me if any of this information is wrong, I'm still learning too.

Patchlead, 2021-02-05:
I have a similar question to this post, so I’ve added it here. your software is great, but i have hit a wall. I have a Large snake enclosure and the temp environment settings work well for the hot area. But, I need to monitor a cool area and a hot plate as well.

- I have a area heater hooked to the temp environment area and on the average temp. (this works well)
- I need a fan hooked to the cool area temp (to turn on if that temp sensor gets too warm)
- I have a hot tile for basking which needs to be linked to its own temp sensor. This is because it needs to be hotter than the area temp, and if i leave it on it can get too hot.

I'm not using PWM heaters or anything, just turn them off and on.

Is this something that can be done? you have a great interface and the background frame work is good so i'd love to use it, but i do not want to run multiple systems on the same tank. All it would need is the ability to duplicate the temp environment setting. and each one can be set to different input temp sensors and output switches

Thanks
Phill

theyosh, 2021-02-05:
@Patchlead at the moment this is not possible. It is a design flaw in the current version. I am working on a new version, where it is possible to have multiple heating and cooling area's. They do work independently from each other.

Patchlead, 2021-02-05:
Thank you for the quick reply.

I have been looking through the code, and it would have taken ages to grow
to the current level, I understand. Whilst I'm not good at python, I've
used other languages.

Is there a section of your code I could look at that I could inject a quick
and dirty "if sensor = x set switch y" statement directly? There is way to
much code for me to easily get it all sucked in to my head at the moment..

WorriedTitan, 2021-02-06:
Elsewhere in the readme it says that decisions are made on the average value of the sensor readings, so if you have 5 temp sensors, it will decide what to do based on the average temp of all 5. 

I imagine that you will need to somehow undo this as well. It seems you can trigger multiple switches off of a sensor reading, but not multiple switches off multiple readings

theyosh, 2021-02-13:
@WorriedTitan Euh, not sure, but if you have 5 temperature sensors and 3 switches, my software assumes the sensors are 1, and the switches are 1. So yes, I can just toggle 3 switches on the average temperature alarms of the 5 sensors. That is how you setup it up yourself.