theyosh/TerrariumPI #495: Has anyone been able to successfully update the first time?

cyber-shifter, 2020-12-27:
Raspberry Pi 4, 3.9.8 previously installed. I followed the instructions here to do the update. The only difference between what I did and the guide, was instead of restarting Terrarium Pi, I just let the install restart it for me when I was prompted at the end of the process. Did this on two machines, both failed for two tries. I suspect one may have a card that is too big as it hasn't ever really worked right, but the second one was in a working state when I updated it. When it booted back up, the MoTD hadn't changed the version, and I couldn't get to the interface at all. Is it normal for the update process to not work the first couple times you do it? Also, I can't find the service running under the name you have listed in the wiki, unless it is hidden in a way I don't know of.

captianboat, 2020-12-27:
i have same issue :(

cyber-shifter, 2020-12-27:
@captianboat I'm willing to bet that a wipe and reinstall will work. But we shouldn't have to do that for every new version. I'll try it later.

kebs19, 2020-12-27:
Same here.... tried a couple of times but I keep getting this semaphore issue...

Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 3, in <module>
    monkey.patch_all()
  File "/usr/local/lib/python3.7/dist-packages/gevent/monkey.py", line 1222, in patch_all
    patch_thread(Event=Event, _warnings=_warnings)
  File "/usr/local/lib/python3.7/dist-packages/gevent/monkey.py", line 195, in ignores
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/gevent/monkey.py", line 756, in patch_thread
    _notify_did_subscribers=False)
  File "/usr/local/lib/python3.7/dist-packages/gevent/monkey.py", line 443, in _patch_module
    gevent_module, target_module, target_module_name = _check_availability(name)
  File "/usr/local/lib/python3.7/dist-packages/gevent/monkey.py", line 429, in _check_availability
    gevent_module = getattr(__import__('gevent.' + name), name)
  File "<frozen importlib._bootstrap>", line 980, in _find_and_load
  File "<frozen importlib._bootstrap>", line 149, in __enter__
  File "<frozen importlib._bootstrap>", line 88, in acquire
  File "src/gevent/_semaphore.py", line 273, in gevent._gevent_c_semaphore.Semaphore.__enter__
  File "src/gevent/_semaphore.py", line 274, in gevent._gevent_c_semaphore.Semaphore.__enter__
  File "src/gevent/_semaphore.py", line 175, in gevent._gevent_c_semaphore.Semaphore.acquire
  File "/usr/local/lib/python3.7/dist-packages/gevent/thread.py", line 121, in acquire
    acquired = BoundedSemaphore.acquire(self, blocking, timeout)
  File "src/gevent/_semaphore.py", line 175, in gevent._gevent_c_semaphore.Semaphore.acquire
  File "src/gevent/_semaphore.py", line 200, in gevent._gevent_c_semaphore.Semaphore.acquire
OverflowError: Python int too large to convert to C long
2020-12-26 22:48:59,000 - INFO    - terrariumWrapper     - Restart counter: 1/5
2020-12-26 22:48:59,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 11 seconds.

captianboat, 2020-12-27:
i just format sd card raspberry pi and  reinstall still not working :(

cyber-shifter, 2020-12-27:
> Same here.... tried a couple of times but I keep getting this semaphore issue...
>
> Traceback (most recent call last):
>   File "/home/pi/TerrariumPI/terrariumPI.py", line 3, in <module> [...]

I think that is similar to what I saw too. I'll have to go look again.

kebs19, 2020-12-27:
im trying to swap sd cards, but no success at all. it´s a serious bug.
using a pi zero w here.

theyosh, 2020-12-27:
run the following command and it should run again: `sudo pip3 install gevent==20.9.0` and then restart. It appears the latest Gevent has some issues

theyosh, 2020-12-27:
Made a ticket for gevent: https://github.com/gevent/gevent/issues/1733

cyber-shifter, 2020-12-27:
> run the following command and it should run again: `sudo pip3 install gevent==20.9.0` and then restart. It appears the latest Gevent has some issues

Thanks so much for the quick reply! I'll give this a go later today. I'm sure the others will too.

kebs19, 2020-12-27:
hey there, nope didn't work...

ModuleNotFoundError: No module named 'numpy'
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 21, in <module>
    from terrariumEngine import terrariumEngine
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 34, in <module>
    from terrariumWebcam import terrariumWebcam, terrariumWebcamSourceException
  File "/home/pi/TerrariumPI/terrariumWebcam.py", line 11, in <module>
    import cv2
ImportError: numpy.core.multiarray failed to import
2020-12-27 16:55:04,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 77 seconds. Press Ctrl+C now to terminate TerrariumPI.

kebs19, 2020-12-27:
had to run:
sudo pip install numpy
and it now boots... I'll try to restore my previous backup and see how it goes :)

kebs19, 2020-12-28:
It works but a couple of things aren't working anymore:
No MOTD when I ssh the server
And the worst: DHT 11 isn't working anymore. Tried changing gpio pins, reset the system, power down, but no way. Even tried another DHT 11 but no success.
For some reason, here in Brazil a bme280 is almost 4x more expensive than a dht11. So we can't just swap the sensors that easily.
And the BME uses more pins to work, so I would have to change the whole cable+connectors between the controller and the aquarium.

theyosh, 2020-12-28:
@kebs19 You are still running on Python2 ? Because the DHT sensor does have issues, and I changed a python library. 
So, if you setup was running fine before the upgrade, it could be a solution to downgrade to the version 3.9.8. I have to admit, that my releases are not always fully tested, and it could be that the DHT change is not working for you.

But, I have never tried a downgrade. In theory you should be able to do:

`git checkout 3.9.8` and then rerun the installer. This should go back to the previous version. And to make sure all libraries are installed, please rerun the installer. And it can help to rerun the installer multiple times. Sometimes the python libraries are not downloaded correctly.

Another solution could be to upgrade to Python3 if you are still on Python2. https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-use-python-35

I think the first priority is to get the system back up running.

theyosh, 2020-12-28:
@kebs19 about different sensors.

I do not know how you use the DHT at the moment. For only temperature use the 1-wire DS1820 is a good alternative. It is also waterproof if you have a metallic head. https://www.amazon.com/Gikfun-DS18B20-Temperature-Waterproof-EK1083x3/dp/B012C597T0/

If you need humidity, then BME280 is a good alternative. https://www.amazon.com/Organizer-Temperature-Humidity-Atmospheric-Barometric/dp/B07V5CL3L8/ Yes, it needs 4 wires, and it needs some soldering. But they are much more stable.
It is using the I2C bus, which is also more stable, and faster to use. I used phone cables for those sensors. Here, the phone cables have 4 wires, so they are perfect for me. And you can put your LCD panel on the same cables, as the I2C bus can just be shared with multiple sensors, as long as the sensors are unique.
And to safe your sensor from direct water, I use plastic casings (ex https://www.amazon.com/Abgream-Plastic-Beads-Storage-Containers/dp/B0825PSQ76/ ) and make some small holes in it. And put a wire through. This way, you can still measure temperate and humidity but the sensor should not be hit by direct water.

kebs19, 2020-12-28:
Hey there :))
I'm still using Python 3, but I'll try this workaround of using 3.9.8, because I'm using it in an aquarium and the only use I had for humidity was to trigger the fans inside the aquarium cover to keep the LED lights cool. 
If you're planning to offer multiple zones on 4.0, I'll use more ds18b20's to monitor the temperature in three zones: inside the cover, inside the controller box and in the water. That will solve all my issues for good. 
And I think that I'll get rid from that annoying issue where the sensors stop working from time to time.

kebs19, 2020-12-28:
By the way I'm making a new and improved version of my controller and documenting it with pics, so very soon you'll see it ready and finally looking good :)
(It was made in a wood box glued together with hot glue and appendices of styrofoam and PET bottles 😂😂.... But worked like a charm for almost 2 years!)

theyosh, 2020-12-28:
@cyber-shifter nope!. Just run the pip3 command and restart. That it.

But, in this case, I added the fix also in the installer, and so doing a git pull and rerun the installer would give the same outcome. But takes way more time which is not needed.

cyber-shifter, 2020-12-28:
@theyosh that worked! Cool. I'm still seeing a lot of lag with connections to the unit (SSH/Web) and the MotD never came back. I think I'm going to try a wipe and reinstall. Do you have a max card size we can use? I know this one is 32 GB, the rest of mine are 16 GB.

theyosh, 2020-12-28:
@cyber-shifter there is no max size to use. The minimal is I guess about 8GB. Smaller will give problems.

But strange that is not working smoothly. Are you sure it is not something to do with your network?

cyber-shifter, 2020-12-28:
I have four Pi's running Terrarium Pi, this is the only one I'm seeing this problem on. But not seeing the MoTD also seems to point at an incomplete/botched install. I'm going to check the speed of the card as well, it may be slow.

cyber-shifter, 2021-01-06:
The speed of the card is the same. Different speed class, but same speed. It is the only 32 GB card though. I may get another card, and try that. I can probably close this issue now.

closed 2021-01-08 by cyber-shifter