theyosh/TerrariumPI #496: [BUG]: Interface Can't Be reached after update 3.9.9

captianboat, 2020-12-27:
**Setup:**
 - Python version: 3
 - Raspberry PI: 4

Interface Can't Be reached after update 3.9.9
so i try format SD card and reinstall is still not working 

**Screenshots**
[image]
[image]

theyosh, 2020-12-27:
run the following command and it should run again: `sudo pip3 install gevent==20.9.0` and then restart. It appears the latest Gevent has some issues

theyosh, 2020-12-27:
Made a ticket for gevent: https://github.com/gevent/gevent/issues/1733

captianboat, 2020-12-27:
it's working now 
but i can't save environment setting :(

WorriedTitan, 2020-12-27:
Thank you, this worked for me

theyosh, 2020-12-28:
This issue is fixed with a new Gevent version. But that version has also still a bug with the picamera which is harder to fix: https://github.com/gevent/gevent/issues/1735

So the installer is still fixed on the latest working version 20.9.0

@captianboat Is it also working for you? Can we close this issue?

captianboat, 2020-12-29:
@theyosh it's work Thank you :)

closed 2020-12-29 by captianboat