theyosh/TerrariumPI #497: [BUG]: LCD Display messages aren't translated anymore

kebs19, 2020-12-28:
I successfully updated to 3.9.9 but now the messages displayed in the LCD aren't in Portuguese anymore.
Some messages are half Portuguese half English, which is strange since I translated 100% of the available po file...
Are there any new strings that aren't available in the latest po file?

theyosh, 2020-12-28:
Hmm, not that I am aware of. I have to look into this. But is your version already running again?

kebs19, 2020-12-29:
kind of... I reinstalled the 3.9.8 version using the backup folder (had to install numpy and gevert again, and the DB got  corrupted so the 3.9.8 installer worked in someway I guess) and now it boots but says it's the v3.9.9 and the dht still isn't working. 
(but the LCD is still in english. I guess that I modified the PO files locally and forgot to commit... I'll check it later)
I'll try my last resort: format and start over...

closed 2021-01-03 by kebs19