theyosh/TerrariumPI #498: [BUG]:

captianboat, 2020-12-29:
**Setup:**
 - Raspberry PI: 4]

**Describe the bug**
can't get notifications Telegram bot 

[image]
[image]

theyosh, 2020-12-29:
In the field Username, you should enter a comma separated list of usernames that are allowed to receive the messages. Else everybody could use your Telegrambot.
I am not sure if you need a '@' in front of the username.

closed 2020-12-30 by captianboat

theyosh, 2020-12-30:
If you restart the TerrariumPI software, you have to say something to the Telegram Bot in order to get updates. The Telegram bot is not able to find the user and start sending messages. So, after each restart, just send something to the bot and you will get updates again

captianboat, 2020-12-30:
:) it's work now but i have new problem with Jack Audio 3.5 i Create a new Topic sorry for many many problem