theyosh/TerrariumPI #499: [BUG]: Jack 3.5 Audio no sound  After Set up TerrariumPI

captianboat, 2020-12-30:
First, sorry for many problem 
before Set up Switches in TerrariumPI Jack Audio  3.5 is working fine but after set up switches and reboot 
Jack not working at all when switch to USB Sound Card is working fine when switch back to Jack 3.5 still no Sound at all 

**USB Sound Card is working fine

theyosh, 2021-01-04:
Hmm, that would be my first question. Do you use PWM? So that is a no.

Did you also reinstall the Pi? I guess that the sound is then through HDMI. Use `raspi-config` to change the sound output of your PI.

I do not touch those settings, and in general, it should work. Also, I do not lock/claim the jack output if you do not use the player part of my software.

So I guess this is something outside my software. Try `sudo raspi-config` -> 1) System options -> S2 -> Audio. Do you have a HDMI device connected?

captianboat, 2021-01-05:
I changed setting in sudo raspi-config to jack 3.5 but still no sound at all i try to find what happened by re install pi everything is work but when i install terrariumPi and set gpio for switches Jack 3.5 is not working anymore

theyosh, 2021-01-14:
Hmm, this is strange. And I do not have any clue. This means we need to google for it... 

What kind of Pi are you using? Pi Zero and RPI3B could have issues with GPIO and rights. Not sure if this Jack issue is related

captianboat, 2021-01-16:
i using RPI4 it's very strange.
by the way i try upload some mp3 drag and drop in Audio funtion i just found this 
[image]

theyosh, 2021-01-17:
Hmm, I found this issue also. But after fixing this, there is a new bug with the MediaInfo library. So this takes some more time to fix.

And are you running on Python3?

captianboat, 2021-01-18:
[image]

i running on Python 2.7 :(

theyosh, 2021-01-18:
Euh, that is not correct. I depend how you have installed my software. Did you do:
`sudo ./install.sh 3` or without the '3'? 
Because that depends on which python version you are running.

It is possible to have both python versions installed on your PI.

And I made an update for the audio file uploading. But I still do not have a clue about the jack issue. And I have not a setup to test. So that will take some more time.

But for now do a `git pull` and restart. File uploading should work.
And can you tell if you use the '3' during the installation?

captainboto, 2021-01-19:
yes i do **sudo ./install.sh 3** for install 

i just re-check again i had two python install
[image]

and do **git pull**
[image]

file show up try play with Soundcard, everything work like a charm but when i switch to Jack 3.5 still no sound 
[image]
[image]

so maybe i need to setup headless build for test? now i build Desktop Pi Version

theyosh, 2021-01-22:
Euh... I think that could be an issue. I do not use the Desktop version as it is more software to install and takes more storage. So I advice always to use the Lite version.

But, still, I think this is related to the Raspbarry PI OS. And I can't test this....

captainboto, 2021-01-27:
@theyosh ok, I will test with the Pi2 B model with the Lite version and update you soon as possible.

theyosh, 2021-02-13:
@captainboto Hi, is this still an issue.? Then I would have to say, save the trouble and buy a 10 euro USB soundcard. Saves a lot of testing and hassling...
I would like to close this issue

captainboto, 2021-02-13:
Okay I think so

closed 2021-02-13 by theyosh