theyosh/TerrariumPI #500: [BUG]: How can we contribute to the wiki?

victorhooi, 2021-01-01:
I'd like to contribute to the TerrariumPi wiki - however, I can't see a way to edit, or suggest edits?

Is it possible to open up the wiki for editing, or PRs?

Or is there some other way to collaborate on documentation for the project? =) Would love to give back.

cyber-shifter, 2021-01-01:
I second this notion.

theyosh, 2021-01-04:
I have tried this before with other people. And it is not possible to use the Wiki with PRs. Somehow that does not work.

But I think what could be an option is to use https://docs.github.com/en/free-pro-team@latest/github/working-with-github-pages

That should be able to work with PRs

But then we should think of some kind of design, and how to style the pages.  And I have never used Githug pages. So I do not know what needs to be done to get it working.

victorhooi, 2021-01-05:
Is granting others edit access to the Github wiki an option? If so, that is one way.

If not, there are various Markdown based documentation tools.

[ReadTheDocs](https://readthedocs.org/) is quite good, and I've used it before. It's more designed around libraries/code projects, but it could work here. Your docs will be in a repo, and we can file PRs etc.

Alternatively, how about something like [Material for MkDocs](https://squidfunk.github.io/mkdocs-material/)?

This is used for the https://perfectmediaserver.com/ site, which I just found.

The template looks visually good, it integrates well with Github (you can click the pencil icon on each page to edit), and is backed by a repo behind it. All the pages are just .md Markdown pages.

I'd be happy to copy the existing content into a repo, and set this up, so we can give it a whirl?

se7en9057, 2021-01-06:
what if we started our own forums with built in wiki? I know we want to link it to this GitHub. i also understand its yet another place to mess with stuff but it may lend its self to community contribution and or community support similar to how the issues section is working now. its a thought. I defiantly want to help with setup tutorials and wiring diagrams. also financially as i'm able.

se7en9057, 2021-01-09:
another thought is a Discord server for community chat / help etc.

cyber-shifter, 2021-01-09:
Discord is great, but it can devolve easily. I think it's a great idea for the collaboration effort, and they're easy to admin. I run a couple myself. I definitely would like to see more tutorials and guides to using the system with all the hardware. I too, am willing to contribute labor to this effort, and possibly monetary support.

cyber-shifter, 2021-01-09:
> Is granting others edit access to the Github wiki an option? If so, that is one way.
>
> If not, there are various Markdown based documentation tools.
>
> [ReadTheDocs](https://readthedocs.org/) is quite [...]

That site looks pretty good! Clean cut, with the relevant information presented well. I like how it has the direct link to the repository there too. Do you need a place to host it?

theyosh, 2021-01-14:
Ok, my preference goes to using Github pages for the general information/documentation. Because that we can do inside my code repository. This should be almost the same as ReadTheDocs.

The wiki is not pullable so there is not an option to collaborate.

Discord I have no experience with.

But I would like to say that this could better be done when I have the new version ready. Because some logic has changed and the UI is different. If we make all kind of screenshots now, we have to do it all over again in a couple of months.

So, lets say that everybody that wants to contribute, should subscribe to this issue. When I have the new version ready, I can update this issue so we can work on the documentation.

How about this?

cyber-shifter, 2021-01-14:
> Ok, my preference goes to using Github pages for the general information/documentation. Because that we can do inside my code repository. This should be almost the same as ReadTheDocs.
>
> The wiki is [...]

Wait for the new version gets a vote from me! I'm already subscribed to the whole git.

se7en9057, 2021-01-17:
> 

I'm ok with waiting as well. looking forward to the new build. will be adding to my sensor pile and do more testing.

theyosh, 2021-04-10:
First setup for documentation: https://theyosh.github.io/TerrariumPI/ based on: https://chirpy.cotes.info/ as version 4 is open for some testing: #539 

For now this is done in the 4.x.y.z branch. I think I/we need to think of an index..
- News / Releases
- Features
- Install
- Hardware
  - Sensors
  - Relays
  - Buttons
  - Webcams
- Setup
  - Sensors
  - Relays
  - Buttons
  - Webcams
  - Notifications
  - Settings
- Scenarios (??)
  - Heating
  - Raining
  - 
- Translations
- Contribute
- Donate

theyosh, 2021-10-12:
For everybody how want to contribute to the documentation, here is how: https://github.com/theyosh/TerrariumPI/blob/documentation/docs/README.md

The current Wiki will be there for old TP3 users. For TP4 all documentation is at: https://theyosh.github.io/TerrariumPI/