theyosh/TerrariumPI #501: Problem with old sensor DHT22. Can't Delete

vikrilow, 2021-01-04:
**Is your feature request related to a problem? Please describe.**
When configured my Terrarium PI for the first time, I tested it with DHT22, Them I read some posts with problems with DHT22 and bought other sensor type.
My problem is that when I run Terrarium PI, i can see old sensor in my system. I've deleted sensor ever, but when reload system, i can see old sensor again.

[image]

Please, can you help me with this problem? ThanKS!

theyosh, 2021-01-04:
If you 'delete' it by clicking on the cross icon, that is not enough. You have to press the save button on the bottom to delete the sensor.

If it is still there, a restart is needed.

vikrilow, 2021-01-04:
Thanks.
I've done this, but sensor continue showing in sensors site. 
Maybe I'm not rebooting correctly.
I usually connect to raspberry by SSH and command sudo shutdown... Or reboot. 
I don't know if I must reboot website with other procedure??

theyosh, 2021-01-04:
Strange.

Restarting can be done as easy as: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-restart-terrariumpi

Else, you can also edit the settings.cfg file, and find the DHT sensors there. And then delete the part that is below the [xyz] part.

It could be that it does not delete correctly. But I think that is strange.

vikrilow, 2021-01-04:
hello again.
I've done you told me but problem persist, when reboot Terrariumpi, old sensor show in website.
[image]

If delete sensor in website, in setting.cfg not show. However when reboot system, old sensors show in website and setting.cfg again...
[image]

Maybe installation are incorrect?

vikrilow, 2021-01-08:
Hello
Any suggestion?

theyosh, 2021-01-09:
Hi the sensors you are showing are 1wire sensors. No DHT. So I do not know how to delete a DHT sensor, if there is not any.

I think you are mistaken... You see 1wire sensors. They are automatically added during startup. If you do not want to use them, then you can after the interface is loaded, click on the wrench icon and choice to ignore.

If you do not want to use the 1wire sensors, please disconnect them...

vikrilow, 2021-01-18:
> Any updates?

HI, this morning, in relation with issue #502 I've deleted every switches and my two wires.
Then, I've connected my 2 wires and problem solved.
Thanks a lot for your help and for your project. 

We keep in touch, maybe I'll open other issue hehe.

theyosh, 2021-01-19:
Ok, then I close this issue. Good to hear it is working now.

closed 2021-01-19 by theyosh

vikrilow, 2021-01-21:
Hi @theyosh I've seen today that problem persist.
I'll tried to explain better my case.

settings.cfg only show real installed sensor:
[image]

However, in website show it:
[image]
 I can delete these and save with green bottom, but when I reload Raspberry, Old sensors show in website.

In fact, It's rare

vikrilow, 2021-01-22:
Hello again,
I'm gone crazy, with strangers sensors... 
I've seen that Mi Temperature and Humidity.. is a bluetooth sensor I've in my kitchen 
Really I don't want to configure in my terrariumPI, but when I reboot system, sensor it's here again...
The problem is that I've been observing that sensor conflict after some minutes. (For example, Enviroment show "error" in Temperature and swicthes shut...
[image]

The same problem applies to Humidity too:
[image]

Really I don't know pass it.

If I restart system, run correctly but when some minutes pass, return error.. 

Any suggestion?

vikrilow, 2021-01-22:
Sorry, I copy somes errors..
2021-01-22 09:34:40,893 - ERROR   - terrariumEnvironment - Environment humidity sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2021-01-22 09:34:40,891 - ERROR   - terrariumEnvironment - Environment temperature sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2021-01-22 09:34:40,777 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2021-01-22 09:34:40,777 - INFO    - terrariumAudio       - Update done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2021-01-22 09:34:25,893 - INFO    - terrariumEnvironment - Update done in 0.00519 seconds. Waiting for 14.99481 seconds for next update
2021-01-22 09:34:25,891 - ERROR   - terrariumEnvironment - Environment humidity sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2021-01-22 09:34:25,890 - ERROR   - terrariumEnvironment - Environment temperature sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2021-01-22 09:34:10,877 - INFO    - terrariumEnvironment - Update done in 0.00519 seconds. Waiting for 14.99481 seconds for next update
2021-01-22 09:34:10,876 - ERROR   - terrariumEnvironment - Environment humidity sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2021-01-22 09:34:10,874 - ERROR   - terrariumEnvironment - Environment temperature sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2021-01-22 09:34:10,772 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2021-01-22 09:34:10,772 - INFO    - terrariumAudio       - Update done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2021-01-22 09:33:55,876 - INFO    - terrariumEnvironment - Update done in 0.00528 seconds. Waiting for 14.99472 seconds for next update
2021-01-22 09:33:55,874 - ERROR   - terrariumEnvironment - Environment humidity sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2021-01-22 09:33:55,873 - ERROR   - terrariumEnvironment - Environment temperature sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2021-01-22 09:33:40,859 - INFO    - terrariumEnvironment - Update done in 0.00509 seconds. Waiting for 14.99491 seconds for next update
2021-01-22 09:33:40,858 - ERROR   - terrariumEnvironment - Environment humidity sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2021-01-22 09:33:40,856 - ERROR   - terrariumEnvironment - Environment temperature sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2021-01-22 09:33:40,757 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2021-01-22 09:33:40,756 - INFO    - terrariumAudio       - Update done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2021-01-22 09:33:25,858 - INFO    - terrariumEnvironment - Update done in 0.00527 seconds. Waiting for 14.99473 seconds for next update
2021-01-22 09:33:25,856 - ERROR   - terrariumEnvironment - Environment humidity sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2021-01-22 09:33:25,855 - ERROR   - terrariumEnvironment - Environment temperature sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2021-01-22 09:33:10,842 - INFO    - terrariumEnvironment - Update done in 0.00509 seconds. Waiting for 14.99491 seconds for next update
2021-01-22 09:33:10,840 - ERROR   - terrariumEnvironment - Environment humidity sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2021-01-22 09:33:10,839 - ERROR   - terrariumEnvironment - Environment temperature sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2021-01-22 09:33:10,741 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2021-01-22 09:33:10,740 - INFO    - terrariumAudio       - Update done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2021-01-22 09:32:55,840 - INFO    - terrariumEnvironment - Update done in 0.00517 seconds. Waiting for 14.99483 seconds for next update
2021-01-22 09:32:55,839 - ERROR   - terrariumEnvironment - Environment humidity sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2021-01-22 09:32:55,837 - ERROR   - terrariumEnvironment - Environment temperature sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2021-01-22 09:32:40,825 - INFO    - terrariumEnvironment - Update done in 0.00518 seconds. Waiting for 14.99482 seconds for next update
2021-01-22 09:32:40,823 - ERROR   - terrariumEnvironment - Environment humidity sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
[... cut ...]
2021-01-22 09:31:25,804 - INFO    - terrariumEnvironment - Update done in 0.00688 seconds. Waiting for 14.99312 seconds for next update
2021-01-22 09:31:25,802 - ERROR   - terrariumEnvironment - Environment humidity sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2021-01-22 09:31:25,800 - ERROR   - terrariumEnvironment - Environment temperature sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!