theyosh/TerrariumPI #502: Problems with connections cables. 4 Relay module

vikrilow, 2021-01-04:
**Is your feature request related to a problem? Please describe.**
Hello,
I'm installing terrariumpi for Tortoise.
Id like controller Humidity and Temperature using Humidifier, fans and Ceramic lamp.
So, I need controller them by relay module. I've this relay module:
[image]

This module are connect to Raspberry but I don't know how connects machines (humidifier, Ceramic Lamp, ...)
About fans, these must connect to 5V but I need to connect 5 fans, so I don't know how I must do.

I've review "Pictures of running TerrariumPI system" but don't solve my problems.. 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Additional context**
Add any other context or screenshots about the feature request here.

bggangel, 2021-01-05:
I had the same challenges that you had about a year ago, and I figured it out using these two pages:

http://www.methinksinc.com/raspiviv/main.html    
Goto the RELAYS tab. I used Vivarium before I used TerryPi (TerryPI is way way way better)

and for an alternative page
https://dev.to/alanjc/water-your-plant-using-a-raspberry-pi-and-python-2ddb

With those 2 pages, you should be able to figure it out.

captianboat, 2021-01-07:
Hi, i just drawing simple picture you can modify 220v/110v to Electrical socket 
power in relay you can use 220v/110v or Dc power whatever you like for Fan, Light, Humidifier.

[image]

vikrilow, 2021-01-08:
> Tuve los mismos desafíos que tu tuviste hace aproximadamente un año, y lo resolví usando estas dos páginas:
> 
> http://www.methinksinc.com/raspiviv/main.html
> Vaya a la pestaña RELÉS. Usé Vivarium antes de usar TerryPi (TerryPI es mucho mejor)
> 
> y para una página alternativa
> https://dev.to/alanjc/water-your-plant-using-a-raspberry-pi-and-python-2ddb
> 
> Con esas 2 páginas, debería poder resolverlo.

Hello,
Thanks! First URL don't run :(
I'll check 2nd URL.
Can you send me some picture about your instalation?

vikrilow, 2021-01-09:
> Hi, i just drawing simple picture you can modify 220v/110v to Electrical socket
> power in relay you can use 220v/110v or Dc power whatever you like for Fan, Light, Humidifier.
> 
> Ps. sorry for wire color
> 
> [image]
> [image]
> [image]

Thanks for drawing!!
Really your installation are very complex for me hehe
I've other doubt about how connect machines, for example one humidifier, Have I cut original cable from Machine?

captianboat, 2021-01-09:
Hi, @vikrilow yes you can cut original cable from Machine it's easy for connect but you will lost the plug too!

I prefer connect wire relay to socket is the best way for fixed and maintenance if your humidifier using 110 or 220v 👍   

[image]

theyosh, 2021-01-09:
@vikrilow If you are not familiar with this hardware, I advise you to google for it, or ask a friend. We are working with 220V live current. So I suggest if you have doubts, ask someone in your area to help.

As these GIO switches are common device, so there is enough information on the internet. https://tutorials-raspberrypi.com/raspberry-pi-control-relay-switch-via-gpio/

vikrilow, 2021-01-18:
> @vikrilow If you are not familiar with this hardware, I advise you to google for it, or ask a friend. We are working with 220V live current. So I suggest if you have doubts, ask someone in your area [...]

HI again and thanks for yours help.

I've connected relay module like picture from@captianboat.
Them I've tried relay module with https://tutorials-raspberrypi.com/raspberry-pi-control-relay-switch-via-gpio/ and run, I can switch on/off. 
However, in Terrarium pii can't be able to controller it. 
I've configured in manual mode unsuccessful :(

[image]

Any help is very welcome

Thanks

captianboat, 2021-01-18:
Hi can you tell me about what GPIO you connect relay to Pi board? 

IN1= Yellow wire  > Pi GPIO =
IN2= Orange   > Pi GPIO =
IN3= Brown   > Pi GPIO =
IN4= Dark blue   > Pi GPIO =
i think maybe you put the worng GPIO number you can check form  https://pinout.xyz/# 
[image]

If you can take picture GPIO Pi board and capture Screen TerrariumPi Switch Setting  is very helpful :)

[image]

vikrilow, 2021-01-18:
Thanks Captainboat,
I think there is the problem.
IN1= Yellow wire > Pi GPIO = 27
IN2= Orange > Pi GPIO = 17
IN3= Brown > Pi GPIO = 23
IN4= Dark blue > Pi GPIO = 22

I've configured with GPIO Number instead of pinout number... 
[image]

I've changed N2, for example to 11 number but I can't switch on/off relay. I'll going to delete every switch's and configure again.

vikrilow, 2021-01-18:
Yes, I've deleted existing switches and I've confiure again with pinout number and run correctly, thanks.

Thanks a lot !!!

vikrilow, 2021-01-19:
Hello again,

Next step is configure fan system. I need to connect 3 fans 6x6 cm and 2 fans 8x8.
Really I don't know what fan models I can buy (really what model are better for this system).

I'll connect in other module relay, but I don't sure what PC Power Supply is better for connect 5 fans.

For configurations, there is some doubts.. 
I'd like to configure fans next form:

1) Depending for a sensor humidity:  If humidity is high, so run fans some minutes. (in this case, I think I know configure it)
2) Apart from that, I'd like to configure running fans some minutes hourly.. I don't know if both configurations are possible.. 

Thanks!

captainboto, 2021-01-19:
first you can decide what you need 12v/24v/48v  /for  me 12v DC 

I choose IP68 fans b'cos my terrarium working on high humidity environmental like 85-100% all day and wet 
it's ok to buy non-IP68 fans i don't know in your area 

**Exsample depend on fans spec you need to read on fans label** 
**1 fans for 6x6 = 12v DC 0.80A (12*0.80=9.6w) 
**3 fans =9.6*3= 28.8w****

**1 fans for 8x8 = 12 DC 1.2A (12*1.2=14.4w)**
**2 fans = 28.8W**

**5 fans = 28.8w+28.8= 57.6W**

next you can find a good DC Power supply (12V / >58 W) maybe 75W whatever you like.

PS. i like Meanwell power supply.

I think 2 configure can't use the same switch.
You need to set up 2 switch like 6x6 run hourly set by Timer funtion. And  8x8 for high humidity set in environmental funtion

theyosh, 2021-01-19:
@vikrilow It is correct that is is not possible to have the same switch being used with sensors and a timer.

So that means you have indeed 1 switch for the sensor setup.
And another switch for the timer.

You can off course connect more than 1 fan to the power supply. But in my software, you need to configure at least 2 switches

vikrilow, 2021-01-21:
OK, no problem, I'll configure fans separelly.
About Power supply I've seen model LRS-100-12 (I think same model than @boat)

About fans, I've seen these, but there are 3 pins (3 cables) and I don't know if this model can connect to this system with Power supply LRS-100-12.

Sorry for disturb!!

captainboto, 2021-01-21:
@vikrilow
1. Cut pinhead is an easy way to do connect **Redline to DC+, Blackline to DC-**  Yellow line not need :) if the wire short you can solder the wire 👍

theyosh, 2021-01-22:
I guess the yellow is used to read out the RPM of the fan. That is used in motherboards, but you can just remove it. I do not use the RPM feedback, as I can't read it from the fan to the PI.

theyosh, 2021-01-22:
And as power source you can use: https://www.amazon.com/MEAN-WELL-RS-75-12-Supply-Single/dp/B005T8XCLW/

They are pretty good at doing their job :)

theyosh, 2021-02-13:
@vikrilow is this all working? Can we close this issue?

vikrilow, 2021-02-13:
Yes,  you can close it.
Thanks a lot for your help everybody!!

closed 2021-02-13 by theyosh