theyosh/TerrariumPI #503: [BUG]: Password shouldn't be stored in plaintext!

Benjin, 2021-01-05:
Please salt and hash, and store that instead.  Changing it will likely require a UI, but seems worthwhile!

theyosh, 2021-01-05:
Hi, yes you are right. But when I started this project, I did not have authentication. It was never meant to use authentication.

And I have a small warning that when you want your PI on the internet, secure it yourself. Something like a NGINX with authentication in front.

But, as I am working on a new version, which is a full rewrite, I can change this. For the current version, I will leave it as is. I hope to release a new version within months from now.

Benjin, 2021-01-05:
I'm looking forward to v4!  Btw, is there any way I can donate a beer or pizza or something?  This project has been keeping my plants alive for two years now, and I very much appreciate all the work you're doing.

cyber-shifter, 2021-01-06:
I too, feel this way. I couldn't set up my snake enclosures the way I want without this system.

victorhooi, 2021-01-08:
@theyosh Perhaps you could add a donate or funding link to your main page?

https://docs.github.com/en/free-pro-team@latest/github/administering-a-repository/displaying-a-sponsor-button-in-your-repository

Also - this is the first I have heard of version 4 - super exciting!

Is there any info on what the big new features/changes are?

theyosh, 2021-01-14:
@victorhooi That sponsoring is not working for me. I do not use those sponsor companies.

But I have https://github.com/theyosh/TerrariumPI/wiki/Donations

Some sneak previews are in the latest release page: https://github.com/theyosh/TerrariumPI/releases/tag/3.9.9 (Screenshots are from the new version).

And I try to incorporate as much as possible from all the 'wish' issues.
Most important #202 , #237 , #287 , #472

theyosh, 2021-01-20:
I would like to close this issue, as this is done in the new version. It is not doable to implement this in the current version.

Benjin, 2021-01-21:
Sounds good!

closed 2021-01-21 by Benjin