theyosh/TerrariumPI #505: Fork to test system translations to display and register messages

kebs19, 2021-01-06:
Hey yosh!
I'm trying to get the messages that shows in the display to be in Portuguese, but I realized that I have to tinker with scripts that are not in po files.
Basically the display shows the date/time in English and the %raw_data% string also can't be translated using the po files. So I will try to tinker with the scripts and see what happens.
I want to translate the register logs as well. 
If I fork the project, it will make harder to keep the code in sync with your changes? Or I'll have to update the files by hand in my fork?
I have almost zero knowledge in python, so it's a project to make me learn how it works and hopefully make a complete translated version for Brazilian Portuguese.
Is that to hard to start? I have the will to do it but I'm starting from scratch. 
I have some pics of the controller, hope you appreciate it:

[image]
Left: Controllers using transistors to drive the lid fans, night lights, and sunset lights .
Right: ds18b20's connections with the resistor included inside the controller box

[image]
LCD display at the top, 12v font in the center, a 5v step down with USB ports above, Raspberry zero w at the right. Outlets in the bottom. 

[image]

[image]

[image]
Outlets here in Brazil are very different from the rest of the world, a pic to show how they look like. There's 3 rounding pins, all recessed to prevent electric shocks.

theyosh, 2021-01-14:
Nice pictures.... A lot of wires...

Uh, about the translations. Not everything is translated. The logging is always in English, else I cannot debug when you have your logging in Portuguese. I can't read that :) So I will never make the logging translatable.

Also, if you have no python program experience, I think this is a bit hard. The code is rather technical and I am afraid that there is not much to translate more.

I can look into this a bit more with my new version. But I am afraid that is hard to translate the logging part, which is used in the notification part, and also used for your display.

kebs19, 2021-01-15:
Hey there :)
Actually I don't know Python but I love to translate things, so I thought it would be a nice opportunity to get me to learn Python.
I'll make tons of mistakes, trying to translate strings and things like that, but it looks like an interesting project, what do you think?

kebs19, 2021-01-15:
I'll open a new ticket for a wish about translations to keep things organized, ok?

theyosh, 2021-02-13:
Hi, is this still an issue? I think I can't do much about this. And in the new system, this will be different as well. I would like to close this

kebs19, 2021-02-15:
I'll wait for the new system, closing here

closed 2021-02-15 by kebs19