theyosh/TerrariumPI #506: Having Trouble Setting up Temperature and Humidity in Environment page

Zlo98, 2021-01-06:
I am currently trying to regulate my temperature based on a sensor and humidity with a sprayer based on a timer. I have attached screenshots of how I have it setup if anyone notices what I have wrong. I cant get the temperature dashboard readings to show anything other than zeros.
[image]
[image]
[image]
[image]

Zlo98, 2021-01-06:
I am a dummy and hadn't had the Lights setup

closed 2021-01-06 by Zlo98

theyosh, 2021-01-06:
Do not use the same power switch for both low and high alarm.

Your heater should only kick in when low alarm is triggered. For high temp alarm you could use fans or something to cool it down.
Now it will work both day and night. Is that what you want?