theyosh/TerrariumPI #507: [BUG]: currency in euros instead of dollars

cyber-shifter, 2021-01-11:
**More info**
Please look first at the [FAQ](https://github.com/theyosh/TerrariumPI/wiki/FAQ) or [Known Issues](https://github.com/theyosh/TerrariumPI/wiki/Known-issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - Python version: 3
 - Raspberry PI: 4
**Describe the bug**
Currency for energy use is showing up in euros instead of dollars. It doesn't matter much to me as long as the number is right, but I thought I would report because I noticed it.

**To Reproduce / Installation steps**
I did the recent update to 3.9.9 choosing that I was in the US and Central timezone.

**Expected behavior**
I expected it to be in dollars, but the dashboard shows in euros.

**Screenshots**
If you need screenshots I can get them later. Not home yet.

**Additional context**
If this is a config somewhere then I missed it. But I haven't had to set the currency before, so I wasn't sure. I looked around, but didn't see a setting for that. I mostly care that the number is still accurate with the cost number that I set in the settings, I don't care what the unit next to it is. But others might.

theyosh, 2021-01-14:
Hmmm, not sure, but did it work for dollars? Internationalization is sometimes a bitch... :P

I will look into this, but it is a bit of a minor issue. So I hope to have some time to look into this this weekend.

cyber-shifter, 2021-01-14:
Yeah, it used to have dollars. I'm not bothered by it, I'm just mentioning it as sort of a heads-up. Others might care? *shrug*

theyosh, 2021-02-20:
Ok, after some testing I got the following results:

1. Using language 'en' I get English language with Euro sign
2. Using language 'en-GB' I get English language with Pound sign
3. Using language 'en-US' I get English language with Dollar sign

So that is still working for me. I have tested this in Firefox and Chrome. Make sure you reload the hole page after language change.

theyosh, 2021-03-03:
Any change to test this? I think it works like expected

cyber-shifter, 2021-03-03:
I apologize, life has been hectic lately so I haven't tested. I will tonight. If it's any consolation, the unit I set up to do a warm test has been keeping the enclosure at good temps for weeks now. Solid as a rock!

DarthKegRaider, 2021-03-05:
Thanks theyosh,

Changing to en-US and reloading the browser with CTRL+F5 displays the $ sign for me.  Even though i'm Australian, I am assuming that the price is just multiplying the cost of power and water I inserted so won't really be in US $ anyway.

cyber-shifter, 2021-03-05:
Works as expected. I will close this out now.

closed 2021-03-05 by cyber-shifter

theyosh, 2021-03-05:
> Thanks theyosh,
> 
> Changing to en-US and reloading the browser with CTRL+F5 displays the $ sign for me. Even though i'm Australian, I am assuming that the price is just multiplying the cost of power and water I inserted so won't really be in US $ anyway.

Yes, that is correct. It could be that water price is not correct if you do not use 'liters' as water unit value.