theyosh/TerrariumPI #508: [BUG]: Since updating to 3.9.9, switches have stopped controlling GPIO pins

Benjin, 2021-01-12:
**Setup:**
 - Python version: 3.7.3
 - Raspberry PI: 3B

**Describe the bug**
TerrariumPi no signal sent when toggling switches in web UI.  Hardware is hooked up correctly, as I can use a python script to enable/disable GPIO and lights/water pumps turn on and off appropriately.

**To Reproduce / Installation steps**
I followed the (very clear, easy) steps at https://github.com/theyosh/TerrariumPI#installation.  When I manually run `sudo ./start.sh 3`, it says TerrariumPi is starting as user "pi", not "root" - not sure if that's a problem, but I followed the installation instructions, but it said somewhere that it needed to run as root.

**Expected behavior**
When I click the "power" icon on a configured switch, it changes the state of the associated GPIO.  (e.g. pin goes high or low, relay changes state, associated light/pump turns on/off).

**Screenshots**
This screenshot is after toggling several different switches on and off.

**Additional context**
* `screen -r` not displaying any message - error or otherwise - when I change the toggle the switch state in the UI.
* Here's an excerpt from settings.cfg with the switch definitions
* Here's some working Python test code (what I used to verify the correctness of the hardware and pins)

theyosh, 2021-01-14:
Hi,

this is strange, as I have not changed this in this release (as far as I know).

Did you run your test script as user Pi or as user Root? If your script does work with the Pi user, then there could be an issue with my software. If your script will only work with the user Root, then it is a Raspberry PI issue outside the scope of my software.

But a quick fix could be to change the TerrariumPI folder rights. This is should only be needed when your script also only works with the user Root.

`cd TerrariumPI`
`chown root .`

And then start it again with `sudo ./start.sh 3`  Now your software is running with Root rights. Normally my software drops the rights to Pi when you start. This is based on the user that owns the TerrariumPI folder.

theyosh, 2021-01-14:
And not sure if related. But is your system up to date with the latest OS packages. I had issues here with a Raspberry update which broke my USB devices. A week later, an update fixed it again. Could be that you need to update with apt to get the latest packages.

ModestZombie, 2021-01-22:
@theyosh you are right, I'm betting right now it's an issue with the OS packages. Tonight as I was installing the latest (1-11-21) Raspbian lite image, and on first boot I ran:
`sudo apt update && sudo apt upgrade`
I noticed this was in the feed:
[image]
So, if @Benjin will attempt to run that same code as above, you'll probably find it's been resolved :)

Benjin, 2021-01-27:
Sorry; I haven't had a chance to give this a shot yet.  Will try this week.  Thanks for the replies!

Benjin, 2021-01-31:
I'm an idiot.  The issue was that I had entered the GPIO pin number, not the physical pin number.  It says that in the tooltip, but it's a bit hidden.  Is there a reason it uses physical pin rather than GPIO number?  I think that'd be more intuitive.

theyosh, 2021-02-01:
Well, if you look at: https://pinout.xyz/pinout/pin22_gpio25# what number would you use:
22, 25 or 6? That is not handy. And also, just counting from 1 to 40 is easier on the pins. Else you have to count first, and make a translation from the counted ping number to the BCM number. That is more error prone than just entering the physical number.

And this is the case for more then 3 years :) https://github.com/theyosh/TerrariumPI#gpio-numbering

Benjin, 2021-02-03:
> Well, if you look at: https://pinout.xyz/pinout/pin22_gpio25# what number would you use: 22, 25 or 6?

Personally, I would use 25.  When you use the Pi APIs directly (Python, `/sys/class/gpio/gpio<pin_number>`), they use the GPIO number, not the physical pin number.  When I pick "GPIO" or "GPIO Inverse" from the dropdown, I'm thinking in terms of GPIO, not general pin.

> And this is the case for more then 3 years :) https://github.com/theyosh/TerrariumPI#gpio-numbering

Yes it has!  I've been running TerrariumPi for two years now, so I must've gotten it right originally.  I forgot to save my settings.cfg when I upgraded PiOS, so I had to figure it out again from scratch.

Perhaps it would be a more intuitive design to have the label (currently "Address") change depending on the type of hardware  selected, so it could be "Physical pin number" for GPIO, "Address" for others that's appropriate for, "USB device number" for that, etc.

theyosh, 2021-02-13:
Ok, as the issue is fixed, can we close this? I will think about the field names in the forms.

closed 2021-02-14 by Benjin