theyosh/TerrariumPI #509: [FEATURE]: translated default values for notifications messages

kebs19, 2021-01-15:


kebs19, 2021-01-15:
[image]

Here you can see that part of the interface is on Portuguese (Mensagem, título, serviço) but trigger strings and the default messages are in English.

Using TerrariumPI 3.9.9, on a raspberry Pi Zero W.

theyosh, 2021-01-18:
I made the Trigger names translatable. The input fields for Title and Message are not translatable.  This is because you can fill out your own message in your own language.

Also there is not much documentation about this, so I will leave the %raw_data% as default, as this will give all the options that are available to use in your message. So, use email with %raw_data% to see all the variables that are available. And then use the ones you want.

kebs19, 2021-01-24:
That's wonderful! I'll start to review the code and we talk again soon.
I'm planning to sign up for a micropython course, I think it will help me get a place to start this language. I'm coming from Arduino (and it's C-based language) so it's gonna be quite a big challenge :)

theyosh, 2021-02-13:
Can we close this issue?

kebs19, 2021-02-15:
Sure, closing it.

closed 2021-02-15 by kebs19