theyosh/TerrariumPI #510: [BUG]: BMP280 - Pressure from Dropdown List

cleartechsolutions, 2021-01-18:
- Python version: [3]
 - Raspberry PI: [B]

BMP280 Pressure Dropdown - I read somewhere that its supported but only temp and humidity work of the sensor, is it possible to have the Pressure element

theyosh, 2021-01-20:
Hi,

at this moment, there is no option for this. And I have to say that this will not be added in the current version. I am working on a new version 4 which does have the pressure sensor support and also altitude support.

So, I am sorry to say, but for this type of sensor, you have to have some patients till the new version is ready. I hope withing 3 months from now.

cleartechsolutions, 2021-01-20:
Many Thanks, Looking forward to it!

theyosh, 2021-01-22:
Ok, then I close this issue now. And I hope to have a new version ready within 3 months...

closed 2021-01-22 by theyosh