theyosh/TerrariumPI #511: [BUG]: Tedious Bugs -  Settings Does not Save

cleartechsolutions, 2021-01-18:
Hi Pal.

Firstly brilliant system.

With version 3.9.9 i am having the following issues..

- Weather does not save in settings.
- 9 Sensors added, and Saves Fine, but Attempting to Save Environment takes ages and sometimes having to reboot to save one element? As with the Weather settings in Settings does not save the weather location

 - Python version: [3]
 - Raspberry PI: [3B]

theyosh, 2021-01-20:
Hi,

I would like to tackle one issue at the time. 

So what is the url you use for your weather data? Could you post it here without the appid= part?
Do you get an error back in the interface? And what is happing in the screen session: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

cleartechsolutions, 2021-01-21:
http://api.openweathermap.org/data/2.5/weather?id=524901&

It says data saved but if reload page the form item is blank again.
same with enviroment, sometimes it does not save and have to restart.

2021-01-21 18:35:18,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.4.240:8090 ...
2021-01-21 18:35:53,268 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device! If this is not correct, try 'i2cdetect -y 1' to see if device is connected.
2021-01-21 18:35:58,348 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
No GEMBIRD SiS-PM found. Check USB connections, please!
2021-01-21 18:36:28,000 - INFO    - terrariumWebserver   - Running webserver at :::8090
2021-01-21 18:37:00,771 - WARNING - terrariumSensor      - Measured distance value 74.54026cm from hc-sr04 sensor 'Water Tank Level' is outside valid range 0.00cm - 65.00cm in 2.00884 seconds.
^Z^Z^CKeyboardInterrupt
2021-01-21T18:39:24Z
2021-01-21 18:39:24,269 - WARNING - terrariumI2CSensor   - Error closing veml6075 sensor 'UVA'. Error message: [Errno 9] Bad file descriptor
2021-01-21 18:39:24,272 - WARNING - terrariumI2CSensor   - Error closing bme280 sensor 'Top Humidity'. Error message: [Errno 9] Bad file descriptor
2021-01-21 18:39:26,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 248 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3 2 1 restart!
2021-01-21 18:39:36,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.4.240:8090 ...
2021-01-21 18:39:52,307 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device! If this is not correct, try 'i2cdetect -y 1' to see if device is connected.
2021-01-21 18:39:57,378 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
No GEMBIRD SiS-PM found. Check USB connections, please!
2021-01-21 18:40:24,000 - INFO    - terrariumWebserver   - Running webserver at :::8090
2021-01-21 18:41:03,706 - WARNING - terrariumSensor      - Measured uvi value 0.09879067406249993 from veml6075 sensor 'UV Index' is erratic compared to previous value 0.0611178343749999 in 0.00023 seconds.
/usr/local/lib/python3.7/dist-packages/gevent/pywsgi.py:902: ResourceWarning: unclosed file <_io.BufferedReader name='/home/pi/TerrariumPI/log/terrariumpi.log'>
  for data in self.result:
ResourceWarning: Enable tracemalloc to get the object allocation traceback
2021-01-21 18:43:37,757 - WARNING - terrariumSensor      - Measured uvi value 0.1005394615624999 from veml6075 sensor 'UV Index' is erratic compared to previous value 0.05764913937500002 in 0.00024 seconds.
2021-01-21 18:44:35,300 - WARNING - terrariumSensor      - Measured uvi value 0.10169124812499994 from veml6075 sensor 'UV Index' is erratic compared to previous value 0.05620879499999999 in 0.00025 seconds.
2021-01-21 18:44:58,981 - WARNING - terrariumSensor      - Measured uvi value 0.10562494874999992 from veml6075 sensor 'UV Index' is erratic compared to previous value 0.05620879499999999 in 0.00014 seconds.
^CKeyboardInterrupt
2021-01-21T18:45:20Z
2021-01-21 18:45:20,114 - WARNING - terrariumI2CSensor   - Error closing veml6075 sensor 'UVA'. Error message: [Errno 9] Bad file descriptor
2021-01-21 18:45:20,117 - WARNING - terrariumI2CSensor   - Error closing bme280 sensor 'Top Humidity'. Error message: [Errno 9] Bad file descriptor
2021-01-21 18:45:21,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 345 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3 2 1 restart!
2021-01-21 18:45:31,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.4.240:8090 ...
2021-01-21 18:45:47,827 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device! If this is not correct, try 'i2cdetect -y 1' to see if device is connected.
2021-01-21 18:45:52,907 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
No GEMBIRD SiS-PM found. Check USB connections, please!
2021-01-21 18:46:22,000 - INFO    - terrariumWebserver   - Running webserver at :::8090

theyosh, 2021-01-22:
Hi,

your weather url is not correct. And yes, I should give a warning back that it is incorrect. But for now use the following weather url format:

https://api.openweathermap.org/data/2.5/weather?q=[CITY],[COUNTRY]&units=metric&appid=[YOUR_APP_ID]

Based on your original url:

https://api.openweathermap.org/data/2.5/weather?q=Moscow,RU&units=metric&appid=[YOUR_APP_ID]

It will give the same info, but I need the longer url for the week forecast. So when you change your weather url, it should work better.

The other messages in your log are normal. So there is nothing in the logs that could help for now.

So I would like to ask to change the weather url. And please confirm that your sensors are working. Then we can look at the environment part.

cleartechsolutions, 2021-01-22:
You can close this issue down, thanks. Once the weather was completed the bugs seemed to iron out?

Thanks for this - now i can finish this little project off!
Such an amazing bit of software!

closed 2021-01-22 by cleartechsolutions