theyosh/TerrariumPI #512: PWM 2A AC Light Dimmer Module by KRIDA PART 2!

cyber-shifter, 2021-01-21:
I'm using the same KRIDA dimmers that @ModestZombie was using in their issue #486 . I'm using the IRF520 profile for the hardware settings. It works great as far as the dimmer working like it's supposed to...except it has a problem limiting. I've set the low alarm to be sensor mode, using the IR Sensor, and a power on time of 300 seconds. If you look at the temp graph, it gets to the low threshold, but then never stops. Previously, when I was using the relays, the system would get up to the temp and then hover around it. So I'm not sure if maybe I have something misconfigured and don't know it? Or if something is wrong. Also, I know I could set an upper limit, but that doesn't really work well when the ambient temperature around the enclosure changes. The limit now would be too hot later in the year.

[image]
[image]
[image]

cyber-shifter, 2021-01-21:
Also, don't pay attention to the actual temps in the limits. The behaviour is the key, the temps are arbitrary until I get it working right.

ModestZombie, 2021-01-22:
So to make sure I'm understanding this, you are expecting the heating panel to heat until it makes it to the low alarm, and then just hover? 

> I've set the low alarm to be sensor mode
We will want to see details of your Environment settings too. Anything that could be interacting with this dimmer, at least. And have you tried viewing the logs (also post them here) to see what it's actually attempting to do? Sounds like this could be happening from your settings. 

may want to enable debugging if you post logs, just in case: 
https://github.com/theyosh/TerrariumPI/wiki/FAQ#enable-debug

cyber-shifter, 2021-01-22:
@ModestZombie yep! Nailed it on the head! This is the behaviour I observed when using relays instead of dimmers. Besides, it's doesn't make much sense that it wouldn't do this, as it's kind of the whole point of the system. I had the system off, so I turned it back on and enabled debug logging. We'll see what it yields after I give it some time.

ModestZombie, 2021-01-22:
I'm assuming you are using the exact same settings as the relay, minus the hardware profile for the relay yeah? If you could post your Environment settings (at least the heater section of the environment page) when you post the logs that would still be helpful. Also, have you upgraded to the new version of TerrariumPi already or are you using a prior build still? If on the newer build, make sure to run `sudo apt update && sudo apt upgrade` as there are updates to the Pi's OS GPIO packages.

I'm curious, did you get the dimming module from Amazon or Tindie? I only ask because the one I bought from Amazon was a dud - It would be on, full on, from the time you plugged it in (the dimming module is supposed to start the device in the off position). and had both LEDs for the 50hz and 60hz on (it's supposed to have one or the other, not both). Looking around, and the seller appears to sell a V3 of the board on Tindie (which has been working great for me). Amazon sells the V2 board.

cyber-shifter, 2021-01-24:
Ok, so. The easy questions first. I got the dimming module from Tindie. It has definitely varied the power level successfully before, so I'm not concerned about that. Also, I'm running 3.9.9, so I think that is the latest version. I've taken a screenshot of my environment settings, and attached the debug log as well.
[terrariumpi.debug.log](https://github.com/theyosh/TerrariumPI/files/5863393/terrariumpi.debug.log)
[image]

theyosh, 2021-01-25:
That is not how it works.

The dimmer will stay at a certain amount of % so that the temperature hovers around the middle of the min alarm and max alarm value. It will only go on, if the current value is below the alarm min. Hence it needs to start warming up. When the current temperature is not below the alarm min, it will not go on.

So, when it is to cold, alarm min is flashing, the dimmer should increase with the value at the field, 'Dimmer environment step'. In your case 5%. Then you should not enter a 'power on' value in the environment part. As this is not working for a dimmer. You should use the settle timeout, so that the sensors can adjust to the new temperature.

If you are looking for a heater that only stays on for 5 minutes (300 sec) and then go off, use a normal relay. A dimmer does not work this way. It will only go down by 5% as this is the 'Dimmer environment step' and then it will go up 5% again for 5 minutes (power on field.)

Also, when you have a zero settle time, it will get to hot. The switch will go on, after it should shutdown after the 5 minutes, due to the fact there is no settle time. This will cause overheating, as it takes time for heaters to heat the area and sensors to measure that.

So, I guess you want to change the Power on time to zero, and the settle time to 300. That means, that when it is to cold, the dimmer will increase by 5% and wait for 5 minutes to see if more power is needed. As soon as the temperature reaches halfway between the alarm min and alarm max, it should stay there. This could mean, that the power can also go down by 5% as the temperature is just above the average value of 87F in your case.

If this is not what you are looking for, then I do not understand the issue. Then more/other info is needed.

theyosh, 2021-01-25:
Also, please show the graph of the temperature sensor Hot side. Not the average temperature. As this is not the value that is used.

cyber-shifter, 2021-01-25:
AH, I did not realize they worked different than the relays, with staying in the middle. I will adjust the settings as you suggested, but I didn't observe the temperature doing that in the graph. I have attached a screenshot of the graph. It is interrupted because I turned off the unit, since it doesn't have animals in it yet. I'll adjust the settings and see what happens. I can set a min and max because there are recommended values for that, so that isn't a problem. 

[image]

cyber-shifter, 2021-01-25:
I think five minutes is going to be too long of a settle out though. I may have to play with that. This is because radiant panels do not actually give off heat, but radiant energy similar to the sun. So when the panel turns off, the temp drops. They don't heat the air, which is why I have to use an IR sensor, instead of a normal temp sensor. I'll keep adjusting that value, until I have one that doesn't result in a wave pattern for the temp readings.

cyber-shifter, 2021-01-25:
So I swapped the settings around like you suggested. "Power on" at 0 and "settle timeout" at 60. The dimmer doesn't get any higher than 6%. It resets at every timeout period. So that isn't going to work. lol!

cyber-shifter, 2021-02-03:
Sorry it took me so long to get back to this. I've had other things going on that needed attention. I'm doing some more testing, so I'll post again, but I'm not observing the dimmes acting like you previously said they would @theyosh. My dimmer hits the low alarm point and goes over it, then shuts down to the configured minimum off value (50%). Then, when the temperature goes under the alarm point, it starts building up again. You can see this in the temp graphs and power graphs for the switch.

[image]
 
[image]

theyosh, 2021-02-13:
Ok, that is strange. But the temperature does stay an almost fixed value. That is the goal right? So when you set the lower alarm a bit higher, would it than not be the temperature you want?

And I have changed this in my new setup. There I use a PID like system: https://onion.io/2bt-pid-control-python/

cyber-shifter, 2021-02-13:
@theyosh Yes, that is the goal I wanted here, maintaining a fixed temp value despite the outside conditions. I just arrived at different settings than we first believed would work. Also, I am glad to hear there is improvement coming for this function, because I'm ordering some more dimmer packs to use with other enclosures. Also, I should note, I've never found my dimmer heat sink to get hot at any time with my settings. I will post my settings here later.

theyosh, 2021-11-17:
Closing due to no updates and outdated TP version

closed 2021-11-17 by theyosh

cyber-shifter, 2022-01-11:
Sorry for not closing this, the tail end of my year got very crazy. I'm working on getting the system up to TP4 and then I will post if I have any issues with the new system.