theyosh/TerrariumPI #513: [BUG]: I get a Bright Pi error when updating

EvilDemos, 2021-01-23:
**More info**
Please look first at the [FAQ](https://github.com/theyosh/TerrariumPI/wiki/FAQ) or [Known Issues](https://github.com/theyosh/TerrariumPI/wiki/Known-issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - Python version: 3.7
 - Raspberry PI: 4

**Describe the bug**
I get this error in the installer when trying to update.

fa8al: unexpected line in /home/pi/TerrariumPI/.git/modules/Bright-Pi/packed-refs: # pack-zefs with: peemed fully-peeled sor|ed!────────────────────────────────────────────────────────────────────────┘
      Unable to find current revision in submodule path 'Bright-Pi'

**To Reproduce / Installation steps**
Followed the update guide.

EvilDemos, 2021-01-23:
Also get this when I run `python3 terrariumPI.py`

Traceback (most recent call last):
  File "terrariumPI.py", line 2, in <module>
    from gevent import monkey, sleep
  File "/usr/local/lib/python3.7/dist-packages/gevent/__init__.py", line 86, in <module>
    from gevent._hub_local import get_hub
  File "/usr/local/lib/python3.7/dist-packages/gevent/_hub_local.py", line 101, in <module>
    import_c_accel(globals(), 'gevent.__hub_local')
  File "/usr/local/lib/python3.7/dist-packages/gevent/_util.py", line 148, in import_c_accel
    mod = importlib.import_module(cname)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "src/gevent/_hub_local.py", line 1, in init gevent._gevent_c_hub_local
ValueError: greenlet.greenlet size changed, may indicate binary incompatibility. Expected 76 from C header, got 72 from PyObject

theyosh, 2021-01-26:
Please update your Pi by re-running the install script: `sudo ./install.sh 3`

There where some updates with packages that could fix this. Also, it looks like not all is cloned well from Github.

So, maybe, rename the TerrariumPI folder and start over again with a fresh clone and a install? Rerunning the installer is just fine.

EvilDemos, 2021-01-31:
Ok will try tomorrow when I get back in town

EvilDemos, 2021-01-31:
still getting the same ValueError as above.

theyosh, 2021-02-13:
Sorry for the late reply, but this is strange. I use the same library and brightpi without any issues.

I get the feeling you need to remove the gevent library, update the Pi with the latest software. And reinstall gevent. I do not know if there are files being compiled during installation of gevent.

So try:
`apt update && apt -y full-upgrade`
`sudo pip3 uninstall gevent`

reboot the Pi. My software will crash due to missing library, but that is ok for now
then reinstall gevent
`sudo pip3 install gevent`

and restart once more.... If this does not work, then a complete reinstall of the SD card?

theyosh, 2021-02-20:
Any luck getting this to work? Mine is working with a Raspberry PI 4 and Python3. So it should be able to work.

theyosh, 2021-03-05:
Closing due to no reponse

EvilDemos, 2021-03-06:
I was in the process the other night testing this and im still having the same problem

EvilDemos, 2021-03-06:
I get the same gevent error after a fresh install of the os and starting from scratch

theyosh, 2021-03-12:
Hmm I have no clue... 

So, I want to try once more to see if we can figure it out. What is the current situation of the setup? 
You did a new install? On a clear installed Raspberry PI?

Because it could be a solution to change the gevent version you are using, but first I need in what state your setup now is.

theyosh, 2021-04-10:
Ok, still no clue, but could you try testing version 4? The installation is a bit different, and maybe that will work. #539

theyosh, 2021-07-26:
Any updates? As I cannot reproduce this and therefore, I do not know how to fix

theyosh, 2021-08-02:
Closing due to no idea, and not reproducible.

closed 2021-08-02 by theyosh