theyosh/TerrariumPI #514: [BUG]:

NiklasArbin, 2021-01-24:
**Setup:**
 - Python version: 3
 - Raspberry PI: 4A+

**Describe the bug**
Meross swiches on the same powersupply, but not used in any configugrations are turned off by TerrariumPI. Even switches marked as manual mode are being shut off.

**To Reproduce / Installation steps**
Rename a switch and press save. 

**Expected behavior**
Terrarium should leave the other switches alone and respect the manual control flag.

theyosh, 2021-01-24:
My guess is that you still have timers setup in your Meross app.

NiklasArbin, 2021-01-25:
There are no timers in the Meross app active, I purchased the switch for this purpose only. 

It just happened again, after rebooting the Pi, all switches were turned off by Terrarium Pi. Very visible since switch 5 is the USB which I power the Pi from.

Also, really impressed by your work on this project, great job!

theyosh, 2021-01-26:
Then I am afraid this will be hard to fix. As I do not own this hardware... What is your model type?

NiklasArbin, 2021-01-26:
> Then I am afraid this will be hard to fix. As I do not own this hardware... What is your model type?

mss425f

But, couldn't the terrariumSwitch.py check if the switch is in manual mode before toggling it. Python is not my language but something like `if(isInManualMode && !force) do nothing`

NiklasArbin, 2021-01-26:
From the behaviour, I'm guessing Terrarium Pi retrieves a previoud state from disk(?) and applies it before the meross api has returned it's current state.

theyosh, 2021-02-01:
I guess your layout is different. And I cannot test that. So, I am afraid that I use channel 0 somewhere, where it should be 1. More info: https://github.com/theyosh/TerrariumPI/issues/254

So, than I do not know how to fix this. I have tried in the past with another issue, but that stalled. For me, the only way to add this, is remote access to your PI and Merros.

If you are able to setup remote SSH with port forwarding in your home router, and change your Merros credentials to something simple, I could do some debugging on your PI. 

If you willing to do this, please mail login details to github@theyosh.nl So we keep them safe

NiklasArbin, 2021-02-03:
I think checking if the switch is in manual mode before toggling it automatically would help a lot. Still think this is a race condition between the meross service and local state. Will try and see if I can make sense of it.

Holding of with ssh access for now since my lizard would get very upset if the lights would start blinking on and off.

theyosh, 2021-02-13:
Ok, that is clear. For now, then I can't do anything. I need a working device for testing.
But I am also working on a complete rewrite, so I think that I will try to make it work in the new version. And that we have to leave it as it is now..

theyosh, 2021-03-05:
Then I would like to close now. I am thinking of ordering a ms245e for testing

theyosh, 2021-03-11:
Ok, ordered a ms245e and made it working in my new version. That should be released very soon.

Therefore I close this issue now..

closed 2021-03-11 by theyosh