theyosh/TerrariumPI #515: [BUG]: not ready for python 3.7

keefer223, 2021-01-25:
I've been trying to get TerrariamPI up and running using python 3.7 and after 5 weeks I still can't get it and running. I used both the current release and git clone the master and made little headway.. So, I loaded TerrariumPi into pycharm to help narrow down some of the issues. 

The first issue to over come up is the install script; it appears that the original code was written in python2.7 but unfortunately python3 syntax is different. I used the 3 option after both the install and start and it didn't help. The first indication issues was the syntax for print statements not using python3 syntax. Next not all the python3 modules were loaded as part of the install script, even one module has changed into 2 modules under python3. Which required to manually install the missing modules,  a minor issue; however, it should be part of the install script. 

Using pycharm, python 3.7 I've managed to get the error count down to the following errors.

python errors - 17
xml errors - 74
html warnings - 480+

Sadly to say I didn't keep track of all the changes I've made to get this far. could probable use meld to compare my work against a new git clone.

The concept is what we have been looking for for some time. Currently we are checking the temp, humidity, lights. soil moisture, uv, foggers all manually. For one or 2 terrariums it isn't a problem but for those of us who have more than a half dozen enclosures, it becomes a challenge and time consuming.

**Setup:**
 - Python version: [3.7]
 - Raspberry PI: [4]

**Describe the bug**
it's not really a bug but more of a coding issue trying to write a program that will work with either python2.7 or python3.

**Expected behavior**
I expected to be able to run release 3.9.9 with minor issues.

**Screenshots**

theyosh, 2021-01-26:
Euh, I am afraid that there is something else. I am running this software now for about 6 months with Python3. No problems.

This software is python3 compatible for more then a year now.

So I am afraid you made somewhere an installation error. Fastest way is to make a new SD card, a clean Raspbian OS Lite and start installing my software with the '3' option. `sudo ./install.sh 3`

Because if you did a first install without the '3' option, you have to change manual some stuff https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-use-python-35

keefer223, 2021-01-26:
Ya, I did all that.. in fact I reloaded the sd card twice in case I did
something wrong.but the results were basically the same. I even tried a new
card and without pyCharm, same thing won't run.

I'll save my work and reload the sd card from scratch and see where I end
up. I'll Document each step.

keith

keefer223, 2021-01-29:
I went back to the beginning:

   1. format and loaded the ssd with raspbian lite
   2.

keefer223, 2021-01-29:
I went back to the beginning:

   1. format and loaded the ssd with raspbian lite. I realized that for
   anyone NOT familiar with linux no gui, internet... would be an issue for
   some.
   2. reformatted and loaded the ssd with raspbian with desktop. The
   webinterface and program are running. However what I learned from my
   previous efforts I decided to remove python2.7, which was relatively easy.
   However, upon reboot the program halted after 5 attempts at restarting. The
   error was basically no import module cv2. So I pip'ed opensv rebooted and
   the webinterface and the main program are still running after 16 hours
   3.  I started setting up the environment settings and updated the sound
   card and the system crashed. tried a reboot no go.
   4. I put the sound card back to what it was. the system is back up.

now the bad: because of the previous issue I uploaded pycharm and decided
to inspect the code and to my surprise PyCharm identified some 300 plus
errors; that didn't include syntax warnings or pep8 warnings. Because I
believe the program has its place with our reptiles I plan to keep working
on eliminating the issues pycharm has pointed out, along with documenting
my work. it's an ocd thing.

I have about 2 weeks to work on the code, at which time I'll be going in
for heart surgery. After which I have no idea when I'll be able to get back
to the code. The python issues shouldn't take that long. However, I know
nothing about html /xml perhaps the convalescence will provide me the
opportunity to learn html/xml.

one of the features that I would like to see incorporate into the system is
blynk. I have 8 systems that I plan to run myself. Which means with the
current configuration I'll have to log into 8 different websites to check
the status of all my terrariums.. I believe blynk, using their local server
program, would provide ALL the information at once.

It's surprising how either python 3 is forgiving or pycharm's error
detection accuracy.

sincerely
keith

theyosh, 2021-02-01:
Hi,

I am sorry, but desktop version is not supported. So, if you are not able to work with a Pi headless and SSH, this project is not for you.

I thinks all the problems are related to the fact you are using the desktop image.

also, the software is working fine with python3 for more then 1year. And I am working on a complete new version, so your changes will not be added to my code.

And, yes, I do not follow the 4 spaces indentation nor the max line limits. That is stupid when you work on a really large screen. So if I where you, I would try to learn how to install Raspberry PI headless, with SSH enabled. And then just follow my 4 installation steps.

keefer223, 2021-02-01:
morning,

I can work with headless and ssh, I'm a tech gut, but I'm thinking about
the guys I socialize with and their computer skills; they can turn on the
computer but don't ask them to write any code . For them to try and use the
program without a desktop would be a horror show and a nightmare for me; so
I try to program under the kiss principle.

Surprisingly I've got the program up and running; actually it has been up
for a couple of days. I've worked thru the 300 some python issues except
for 2; osgeo.ogr and osgeo.ogs. GDAL and PROJ 6 have been kicking my butt
trying to build their programs from source code. I finally got proj 6
completed last night, gdal thats another issue.

Believe me, I don't worry about line length limitations or 2 spaces instead
of 4 or the rest of the pep8 guidelines, I just reformat the code before I
dive into the code.

I really didn't expect my changes to be incorporated into your project, I
mean it's your project. All I was trying to do was let you know that there
were some issues, I was working on them. If you wanted to see what I found,
I was more than happy to share not only what I found but what I did to
correct the issue. As I stated earlier I believe this project has merit, so
I'll keep plugging away.

I'm not sure if I'll upgrade anytime soon. As I previously mentioned, I
would like to get blynk incorporated into the program first.

best of luck

theyosh, 2021-02-13:
Euh, I think I cannot do anything about this? My software will not work on a Desktop OS. That does not work well with the GPIO ports. And also, my software is rather technical, because you need to connect all kind of sensors and wires. So if you can only work on a Desktop, then this software is not for them.

Therefore, I would like to close this issue, as my software is working like expected

closed 2021-02-21 by theyosh