theyosh/TerrariumPI #516: How to connect 2 BMP280 sensor

vikrilow, 2021-02-01:
Hello everybody,

I don't know if I can connect 2 BMP280 sensors using same pinouts, like I connect 2 DS18B20 wire sensors.
My problem is that actually I haven't free pinouts to connect 2 BMP280 sensor.
So far I've connected 2 wires DS18B20 (using 3 pinouts only), 2 relay modules and 1 BMP280 sensor.

When I connect to 2 BMP280 sensors using same pinout, Raspberry don't detect them.
[image]

Thanks for your helps.

vikrilow, 2021-02-01:
[image]
I've connected wires in pinouts correctly and detect my sensor, but I don´t know how configure second sensor in TerrariumPI
thanks!

theyosh, 2021-02-02:
Hi @vikrilow 

As far as I know, you should be able to change the address by changing some pins on the board. But I did not had success with that, so I use I2C multiplexing.

More info is here: https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README#L1506

So I added to my /boot/config.txt:

```txt
[all]
dtparam=i2c_arm=on,i2c_baudrate=400000
dtoverlay=i2c5,pins_10_11
dtoverlay=i2c4
```
This will add two new I2C busses. And for bus 5, I changed the default pins.

This way, you can add a second and third BE280 to your system on the new I2C busses. This will also work for other I2C devices that cannot change their address. It is a bit a trail and error.

vikrilow, 2021-02-03:
Thanks for your reply. Really I don't understand very well it. 

Actually, I've connect the sensor on next pinouts:

SDA - 3
SCL - 5
GND - 20 
VIN - 17

So if I configure /boot/config.txt like you told me, where I have to connect the sensor? 
I¡m sorry for disturb, really I'm feeling a stupid hehe

vikrilow, 2021-02-07:
Hello again @theyosh, please, any additional help? 
Thanks in advanced

Dave0408, 2021-02-07:
Good evening @vikrilow , 
this is exactly the topic I successfully implemented this weekend. 
I was able to connect two BME280 in a series connection. They are connected to the same IC2 bus. 
It is important to know that the delivered sensors all have the address 76. But to use two sensors you need two different ones. Here 76 and 77. For this I had to separate a solder connection on ONE of the two sensors with a cutter knife and make a new one with some solder. I got the info for this from this thread: 
https://raspberrypi.stackexchange.com/questions/103962/how-to-connect-two-bme280-sensors-via-i2c-to-a-raspberry-pi-3b

I hope this helps you for the time being. Otherwise, let me know and I'll provide some more info. 

If everything is correct, then it should look like this: 
```
i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- 76 77
```

vikrilow, 2021-02-07:
> Good evening @vikrilow ,
> this is exactly the topic I successfully implemented this weekend.
> I was able to connect two BME280 in a series connection. They are connected to the same IC2 bus.
> It is [...]

Thanks @Dave0408

vikrilow, 2021-02-07:
Thanks @Dave0408
Can you send me picture from.your two sensors?
To see better the differents.. 

Thanks

Dave0408, 2021-02-08:
Hope this helps: 
[image]

Another tutorial link: https://www.bluedot.space/tutorials/connect-two-bme280-on-i2c-bus/

vikrilow, 2021-02-09:
Thanks!
I think I'll need a great magnifier to do it hehe

theyosh, 2021-02-13:
> Thanks for your reply. Really I don't understand very well it.
> 
> Actually, I've connect the sensor on next pinouts:
> 
> SDA - 3
> SCL - 5
> GND - 20
> VIN - 17
> 

If you add the dtoverlay=i2c4 you will get the following pins:
SDA - 24
SCL - 21
GND - 20
VIN - 17

dtoverlay=i2c5,pins_10_11
SDA - 19
SCL - 23
GND - 20
VIN - 17

But it could be that you have to swap the SDA and SCL... I cannot check this with my setup, as that is all closed now. So if the soldering option does not work, you can also try to add more I2C busses to the Pi

vikrilow, 2021-02-13:
Hello
Today I've been modifying sensor. 
Tomorrow I'll configure 
[image]
 it

vikrilow, 2021-02-13:
> If you add the dtoverlay=i2c4 you will get the following pins:
> SDA - 24
> SCL - 21
> GND - 20
> VIN - 17

Hello
I've connect it as you wrote before and run correctly.

[image]

Thanks again for your attention!!

vikrilow, 2021-02-16:
@theyosh any suggestion? 
Thanks!

theyosh, 2021-02-16:
try: `i2cdetect -y 4` and `i2cdetect -y 5`

vikrilow, 2021-02-17:
> try: `i2cdetect -y 4` and `i2cdetect -y 5`

I've tried it. I show you log about these sensors

[image]

[image]

Really, Now I'm very loss :(

theyosh, 2021-02-17:
Hmm, for bus 4 it looks bad wiring. For bus 5, nothing is connected.

It could be that I have mixed up the SDA and SCL pin numbers. So swap those for a try. Also, test with one sensor first, that makes it easier to debug.

vikrilow, 2021-02-21:
Hello @theyosh 
Connecting these pins:
SDA - 24
SCL - 21
GND - 20
VIN - 17
this is result:
[image]

Now Connecting these pins:
SDA - 19
SCL - 23
GND - 20
VIN - 17
[image]

As you can see, connecting 1 sensor only and using clean cables.

This is my config.txt configuration:

[image]

Any suggestions more?

vikrilow, 2021-02-21:
more info from log site:

[image]

vikrilow, 2021-02-21:
I'm sorry for a lot replies... 
I've connected 2 sensors and they show in i2c5:
[image]
However, when I configure in website, they don't run correctly:

[image]

I've rebooting raspberry some times but problem persist...

vikrilow, 2021-02-21:
I'm a Stupid!!
I've configured 76,5 and 77,5 and run correctly... 
I'm using i2c5 do that I've to use ,5 

Now I'm so happy!!!!

theyosh, 2021-02-21:
Nice! It took me also a few attempts to get the extra I2C busses working. But now at least your I2C5 is working. That is nice. And also changing the sensor address is working. Have fun!

vikrilow, 2021-02-21:
@theyosh thanks a lot!!!
Next week I'll open other issue to configure well my particular configuration.. Hehe

theyosh, 2021-02-25:
Your welcome!

closed 2021-02-25 by theyosh