theyosh/TerrariumPI #517: [FEATURE]:Support for BME680

bobnut12, 2021-02-03:
I'm trying to connect a BME680 from sparkfun.  I've installed the drivers from https://github.com/alexh-name/bsec_bme680_linux and have the sensor talking to the pi but Im not sure how to integrate this with TerrariumPi.  My python skill are basic at best.

theyosh, 2021-02-13:
Hi,

this will be a bit more difficult. The output of that script is not suitable for my software. There is also Python code for this sensor, but I am not able to test that. So I suggest to do it differently.

First run: `sudo pip3 install bme680` This should install some libraries. 

Then download an example/test file: `wget https://raw.githubusercontent.com/pimoroni/bme680-python/master/examples/temperature-pressure-humidity.py`

Then run the script: `python3 temperature-pressure-humidity.py` and  see what happens. If this produces some valid outcomes, we can tweak this script to produce valid output for my software.

bobnut12, 2021-02-14:
I've done the steps above, but it appears that the pimoroni libraries don't calculate eCO2.  This is for a mushroom tent and I was hoping to track my eCO2 values. I'll have to rethink my design.  Thanks for your help.

theyosh, 2021-02-19:
Where are the eCO2 values in the other linux script? In the example output I do not see that....

And when you change the temperature-pressure-humidity.py and use the value 'gas_resistance' in stead of 'humidity'. Does that give the value you are looking for?

bobnut12, 2021-02-19:
Using the drivers in my original post, this is the output I receive.  

pi@raspberrypi:~/bsec_bme680_linux $ ./bsec_bme680
2021-02-19 20:29:56,[IAQ (1)]: 57.10,[T degC]: 15.28,[H %rH]: 87.98,[P hPa]: 1014.38,[G Ohms]: 97747,[S]: 0,**[eCO2 ppm]: 556.096374511718750**,[bVOCe ppm]: 0.6180703043937683105468750

After more reading, I found that the IAQ, eCO2, and eVOC functions of this sensor are only reliable when using the bosch drivers. Unfortunately, the Pimoroni scripts dont use these (a licensing issue as I understand it), and are limited to raw outputs.

theyosh, 2021-02-19:
Hmm, and the original code does not have much options. And it keeps running in a loop. So that is not handy. There are ways to work around it, but that would involve a lot of coding.. :( 

So I am afraid that I cannot support this sensor at this moment.

bobnut12, 2021-02-19:
No worries, thanks for your help.

closed 2021-02-25 by theyosh