theyosh/TerrariumPI #518: Updating notification settings failes

kahuwi14, 2021-02-06:
Dear yosh,

I hope, you're doing great!
Yesterday, I reinstalled my Pi3b, because the update process to 3.9.9.
That wasn't a big deal, because it forced me to do what I wanted to do for a long time. ;)

Now, everything is fresh and up-to-date, without errors and strange behaviour, except the issue, that I can't make any setting for the notification!

Without a notifications.cfg, I'm able to fill the form with all needed information, but after saving, all values in the messages-section "jump" one box to the left. That happens after every saving. It doesn't make any difference, if I change a value in this section, or not.

[image]

I tested this Issue of course in the incognito-mode as well.

It wouldn't be a problem, if I would be able to activate push-notifications over the config-file, but that's not working.. :-/

cya
Karsten

theyosh, 2021-02-13:
Sorry for the late reply. I have no clue. I have to look into this. I think I get the issue here. But I need to do some testing. I will get back on it

theyosh, 2021-02-19:
Found it. Do a `git pull` and restart. It should be fixed.

kahuwi14, 2021-02-19:
Great, it works!
Thanks a lot!

closed 2021-02-19 by kahuwi14