theyosh/TerrariumPI #522: [Question]: Start action from environment settings with time delay

Dave0408, 2021-02-10:
Good morning, 

I have another question about the environment settings.
I have set to control the humidity, that at a threshold of 98% for 10 minutes the fans turn on. Now I spray at fixed times (morning 20 sec and noon 30 sec). Immediately after spraying, the humidity is of course extremely high. The fans react to this by starting immediately and pulling the generated humid air directly out of the terrarium. This doesn't make much sense to me. 

Is there a possibility to start the action caused by exceeding a threshold with a time delay?
Example: spray 20 sec -> humidity rises to 100% -> wait 10 minutes -> fans go on.

The only idea I have is to set the fans to manual mode for 10 minutes after spraying using an external pythonscript.
But this is not really comfortable and would be again a point of control outside your software. 

Thanks in advance.

Dave

ModestZombie, 2021-02-12:
@Dave0408, if you can post a screenshots of your settings (humidity Environment settings, Power Switch settings, etc) folks around here may be able to help better. Also, I'm assuming the mister and the fan are connected via relay? I'm no expert at this, but from the sounds of it, you would want to set the humidity Environment "Power on" in the alarm section to a pretty short time frame (like maybe 3 seconds), and set the "Settle timeout" to be 600 seconds. The idea is once Humidity hits your alarm percentage, the fan will still kick on briefly, then wait 10 minutes to see if it needs another short gust of air. Considering it sounds like you want to have high humidity pretty much all the time, you aren't going to want that fan on very often. 

I'm not sure about your setup and what it is your using this for, but just some thoughts - are you able to try flipping the fan around and changing the airflow? Maybe pull dry air IN instead of pulling that Humid air OUT? You may be able to control the humidity better that way. Depending on the vents in your enclosure, you wouldn't see such drastic change in Humidity levels. Thats my theory at least. Hope any of this helps, and I apologize in advance if I'm misunderstanding your question.

Dave0408, 2021-02-12:
@ModestZombie 
Thank you for the detailed answer. 

It is correct that the fans are connected via a relay. 
As fans I use 2x Noctua NF-A9x14 PWM. I pull the air from the terrarium to make the most of the chimney effect through the ventilation under the front window. If I were to blow the air in from above then I would nullify this. 
Also true is that I basically have a high humidity. Always between 80 and 100%. Due to the more frequent ventilation my orchids and mosses develop quite well. 
Basically, the idea of initially a very short phase of ventilation and a longer break is quite good. But in the long run it provides too little airflow. 
My concern is more about giving the sprinkler system a chance. Leaving the plants in the mist for a short time and only regulating the humidity again after a break.
I hope I have expressed myself understandably.

I had documented my setup here:
https://github.com/theyosh/TerrariumPI/issues/210#issuecomment-748327920

And my configuration:

[image]
[image]

theyosh, 2021-02-13:
Euh, this is not possible. You are spraying with a relay with timer? So not in the environment part?

Then yes, when it get to humid, the fan will kick in. But what you want is that it should stay for a certain amount of time at the alarm level before the fan kicks in. Because there is no relation between a single switch and the environment parts.

Or, you need a setting where you can say how long the alarm should be on, before action. Like 1 minute. When you have an alarm for more then 1 minute, the relays (and your fan) will kick in.

At the moment, this is not possible, and I am rewriting the software so I can have a look to get this in the new version. But in the current version, I will not add more features. Sorry for that.

theyosh, 2021-02-25:
Hmm, maybe it is possible. I think I found a way.

I only used the humidity environment part for this. So no timers on the switches them self.
The first part is to use a timer setup and no sensors for this. And configure the low alarm (sprayer)
[image]

Here I changed the mode to timer. So all actions are now done based on time. It does not matter what the actual humidity values are.

Then I chose the start time of 9:00 in the morning to start the spray timer, until 4 in the afternoon. In this period I defined a time slot of 1 minute to be on, and 409 minutes to be off. This should result in a schedule where it will be on at:
 - 9:00
 - 15:50 (9:00 + 1 minute + 409 minute )
So you have no two times a window where the switches are turned on, and would spray water.

Then the nifty part of the fans... That is done on the same way, but with 10 min delay in the timer values:
[image]

Here I have shifted the timer times with 10 min. So this runs with a 10 min delay on the low alarm (sprayer). And here I just set a time slot of 10 minutes, as your fans run more then 1 minute. And a off time of 400 minutes. This should give the following time stamps to go on:
 - 9:10
 - 16:00 (9:10 + 10 minutes + 400 minutes)

And this way, you should have a setup where there is two times a day a spray action of 20 seconds, and 2 times a day a fan action of each 10 minutes.

But this is not tested by me as a setup. So you could try this. Maybe an advice to backup your settings.cfg file first, if this does not work well. Then you can easily revert back to your old setup.

theyosh, 2021-03-03:
@Dave0408 did the above suggestion work for you?

Dave0408, 2021-03-03:
Good morning @theyosh 

thank you for this idea/approach. I have not implemented it yet, because then I would not react to any thresholds at all outside the defined times. I think that is also not in my sense.
But the topic has no hurry. Maybe there is a way with the new software version.

theyosh, 2021-04-21:
He @Dave0408,

how about this:

the settle time out is also enforced when you go from a low alarm to a high alarm. And vise versa. 

The idea of the settle timeout is that an alarm will not trigger more then once in that settle timeout. But there is no relation between the low and high alarm. So, we could also say, that when the low alarm is triggered, the settle time is then also set to the high alarm. And that will cause the high alarm get a timeout of the settle time of the low alarm.
And going from high alarm to low alarm, will then get the settle timeout of the high alarm.

Not sure if this will fit, as you could say, I want a settle timeout of 5 min for the low alarm, but a 10 min settle time between low and high alarm. That will not work. But I think this could be another solution.

I hope it is clear what I mean. And can you think if this will work for your setup?

theyosh, 2021-08-30:
Sorry for the late reply, but I think there is a small mistake. What I suggest is:

The sprinkler system starts -> wait the settle timeout -> humidity still higher then 98% -> fans start

But this only works between the high and low alarm of the same area type. And I think this is pretty 'normal'/'expected' behavior. If you act on low alarm, which will directly trigger the high alarm, is a bit useless.

I think this is the solution for you. But this will be then implemented in TP4.