theyosh/TerrariumPI #525: [FEATURE]: NeoPixel Support

Blacky2812, 2021-02-28:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Additional context**
Add any other context or screenshots about the feature request here.

theyosh, 2021-03-05:
Sorry, but I do not see how to support a NeoPixel, as I will not make a color system to control colors.

Blacky2812, 2021-03-05:
Okay, how i can access top external Scripts? so i can write my own Python top Control ws2812b

theyosh, 2021-03-06:
An example can be found here: https://github.com/theyosh/TerrariumPI/blob/master/contrib/external_switch.py

theyosh, 2021-03-12:
Hi, is the example script enough for you to get started?

Blacky2812, 2021-03-12:
Yep, i got it running.

closed 2021-03-12 by theyosh