theyosh/TerrariumPI #526: Problems to setup a Time(Tp-link kasa plug)

17Trust09, 2021-03-12:
Hey there :) 
not sure if thats the right way to ask for help.

im trying to setup a timer (power plug) but its totally freaking out at the moment (going  on/off again).
any idea why?  my intent is to have it 12h off and 12h on....

[image]
[image]
[image]

theyosh, 2021-03-12:
Only fill in the Tag/Nacht part when it needs to be on. I guess that for you it is the easiest to clear the Nacht (night) part. That should fix it.

theyosh, 2021-03-22:
Any updates? Did the change work?

17Trust09, 2021-03-22:
well I totally forgot to answer  :) 
it's working fine now! I did the timer setup over the timer setup in the Switch tab :) 

love the app so far! next thing that I will try is to setup a temp and humidity sensor.

theyosh, 2021-03-24:
Nice, then I close this issue now. Have fun! And do not forget #210

closed 2021-03-24 by theyosh