theyosh/TerrariumPI #527: Using the same switch for low and high alarm

fabi2803, 2021-03-12:
Hello,
first of all, thanks for working on this great project, I love it and I'm going to use it for my mushroom grow room. I'll send you pictures in a few weeks when I put them in the tent.

Now to my little problem:
I'm using a ultrasonic humidifier with a fan to control the humidity, both with their own switches.
With low alarm both should power up, with high alarm I want just the fan to blow.
[image]

[image]

At first I didn't have a high alarm and it worked fine as you see at the middle of the graph.
As I added the Fan at the high alarm, it turned on and off all the time when humidity was to low.

[image]

I could minimize the problem with a settle timeout at the high alarm setting but I don't think that's a solution.
I don't know how much sense my high alarm strategy makes but maybe there is a scenario where a switch is used at high and low alarm. 

Best regards

theyosh, 2021-03-22:
Hi,
at the moment it is not possible to use a relay twice. So if a relay is selected in the low part, you cannot use it in the high part. That will not work.

In my new version it will work. So for now, you could add a second fan and use that for the high alarm settings.

fabi2803, 2021-03-22:
Do you want to reveal when the new version might go online?

theyosh, 2021-03-22:
The new version will be online (I hope) this month. But it is a major rewrite, so I call it beta. And it can have (new) bugs.

fabi2803, 2021-03-22:
Nice!
I am looking forward testing it.

closed 2021-03-22 by fabi2803