theyosh/TerrariumPI #528: PWM AC Dimmer(YYAC 3S 220V)

steveava, 2021-03-17:
Hi,

Have a couple of these arriving next week.

https://www.aliexpress.com/item/1005001369038922.html?src=google&albch=shopping&acnt=494-037-6276&isdl=y&slnk=&plac=&mtctp=&albbt=Google_7_shopping&aff_platform=google&aff_short_key=UneMJZVf&&albagn=888888&isSmbAutoCall=false&needSmbHouyi=false&albcp=1622677378&albag=57224001810&trgt=743612850714&crea=en1005001369038922&netw=u&device=c&albpg=743612850714&albpd=en1005001369038922&gclid=Cj0KCQjw0caCBhCIARIsAGAfuMzwF55hnwgomw5o81c3L5Aw_oWkD0e6opWpiwAejfES2CJEluim2n4aAgftEALw_wcB&gclsrc=aw.ds

Are there any existing PWM device setups that I could try in TerrariumPI.

Steve.

theyosh, 2021-03-18:
Hi,

I think using as a 'DC Dimmer' should work. Even if you want to use AC.

AndrejWeber, 2021-03-18:
Hi,
If you want to use it with AC-mains, you have to be very careful! Do not touch the module or even the raspberry pi, while mains is plugged in!
It is absolutely not isolated, like a comment tells on Amazon. 
It should be a cut trough the PCB along the white line, but it is only a printed white line. 
So it is a very dangerous peace (of crap)!
It also could burn your raspi. 
I cannot imagine, that it could dim light bulbs without flickering, because it has no zero crossing detection.
A old simple light bulb could work on this dimmer, halogens and LEDs should not.

steveava, 2021-03-18:
Great, I'll try the 'DC Dimmer' see what happens.

The plan was to use either 240VAC dimmable lamps or heating elements. Would be nice to use proportional heating rather than relays and as TerrariumPI uses PWM for other dimming devices, perhaps that may work with this device as it's cheap. Might be a reason for it being cheap!!!

I think this a fairly new device from China but could be wrong and I realise that there's a bad comment on AmazonUK but I have also seen opposing views elsewhere so curious and bought two, one to test with a scope and pull apart, the other to use if it looks safe. I wont be blowing up any Pi's, I'll use a 555 timer to test.

Not sure what you mean by "cut along the white line". The white line looks like it's identifying the "separation" between Mains HV and LV control but as you say there's little documentation. And there appears to be an opto-isolator on the LV side which should protect the control circuit, Pi, Arduino or whatever. Its hard to be sure without having a schematic or one to play with.

Looks like it uses a 8051 micro so will be interesting to see what that's up to. I've used micro's for zero crossing in thyristor control circuits so I suspect that's what it's doing.

steveava, 2021-03-25:
This AC dimmer seems to work ok under test conditions. The HV and LV circuits are optically isolated but the PCB could have been designed better to separate HV/LV tracking. 5v PWM at a frequency of 220hz seems to eliminate flickering although my interest was heating so not that much of an issue. Dropping below 20% duty cycle(including 0VDC) seems to give an erratic output but at 20% you're only getting chopped 8VAC anyway. From 20% to 99% seems to be fairly liner although there is a small but noticeable jump from 99% to 100% or 5VDC.

I went through TerrariumPI DC dimmer setup but couldn't find an option to adjust frequency. Where would that be set.

theyosh, 2021-07-26:
Sorry, missed your remark about frequencies. If you use TP4, you can press the calibrate text to enter the frequency you want.

theyosh, 2021-08-13:
@steveava Did you make any progress?? Did you try TP4? There you can change the frequencies

Else I would like to close this issue, as I can't do much.