theyosh/TerrariumPI #529: [FEATURE]: BM1750 Lux Sensor

wolli1234, 2021-03-23:
Hello theyosh,

thank you so much for this software, I'm setting up a gourmet mushroom microfarm right now, and will post some pictures, when it is ready :).  

I would like to add a light sensor and due to [this post](https://wolles-elektronikkiste.de/sensorvergleich) the BM1750 seems to be a reliable one.

I can read the lux value in the terminal by using this script:

`#!/usr/bin/python3
import smbus
import time

DEVICE     = 0x23
POWER_DOWN = 0x00
POWER_ON   = 0x01
RESET      = 0x07
bus = smbus.SMBus(1)

def convertToNumber(data):
  result=(data[1] + (256 * data[0])) / 1.2
  return (result)

def readLight(addr=DEVICE):
  data = bus.read_i2c_block_data(addr,0x20)
  return convertToNumber(data)

def main():
  lightLevel=readLight()
  print (format(lightLevel,'.2f') + " lux")

if __name__=="__main__":
   main()
`

Would it be possible to add this sensor to TerrariumPi? In the adding sensor section I can choose "Custom script" but I can't find any documentation how this works. 

I'm a total beginner with such RaspberryPi projects and python, would be great if you could help me or put me in the right direction.

Cheers,
Wolli

theyosh, 2021-03-24:
Hi,
if the above script is working for you, you are about 90% there. You need to change the script a bit. The line with ` print (format(lightLevel,'.2f') + " lux")` should be changed with ` print (format(lightLevel,'.2f'))` without the ' lux' part. So it will only output a number when run.

Then you can add a script sensor. The address would then be `/full/path/to/your/script.py` And select light type.

That should do it.

wolli1234, 2021-03-24:
Hi,

thank you, great that works :)!

Just for curiosity. Could you point me to the code where the repeat rate for measuring the value is in TerrariumPi? I thought I would need to add this to my script, but seems like TerrariumPi will update sensor data automatically.

theyosh, 2021-03-24:
No need for repeating stuff. You only need to provide a script that does 1 measurement and print out the number. That is all.

Everything else is handled by the software. My software will take a measurement every 30 seconds. And therefore, execute your script every 30 seconds. That is all.

closed 2021-03-24 by wolli1234

fabi2803, 2021-04-01:
I get in the Terminal the number but the software says "NoneLux"

> pi@raspberrypi:~ $ python /home/pi/BH1750.py 
> 68.33

> 2021-04-01 18:18:16,379 - WARNING - terrariumSensor      - Measured light value Nonelux from script sensor 'Licht' is outside valid range 0.00lux - 100.00lux in 0.06532 seconds.
> 2021-04-01 18:18:16,313 - INFO    - terrariumSensor      - Loaded script light sensor 'Licht' on location /home/pi/BH1750.py.

wolli1234, 2021-04-01:
Just for the record, this is the script working for me:

```

#!/usr/bin/python3
import smbus
import time

DEVICE     = 0x23
POWER_DOWN = 0x00
POWER_ON   = 0x01
RESET      = 0x07
bus = smbus.SMBus(1)

def convertToNumber(data):
  result=(data[1] + (256 * data[0])) / 1.2
  return (result)

def readLight(addr=DEVICE):
  data = bus.read_i2c_block_data(addr,0x20)
  return convertToNumber(data)

def main():
  lightLevel=readLight()
  print (format(lightLevel,'.2f'))

if __name__=="__main__":
   main()

```

fabi2803, 2021-04-01:
Tried your script, still same problem...

This are my settings:
[image]

How are your settings? Or do you see what might be my mistake?

wolli1234, 2021-04-01:
Hmm my settings are the same like yours except I'm using Max diff "0".

fabi2803, 2021-04-02:
> Did you put ADDR to ground?

Yes
 As I said, I'm getting results if I run the script on the terminal.
I even tried float() so that it's a "real" number.

BTW: I'm using this for a mushroom farm, too ;)

wolli1234, 2021-04-02:
The only difference is my script path: /usr/local/sbin/bh1750-smbus.py

And because of a CCS811 sensor I was playing around in /boot/config.txt and added "dtparam=i2c_arm_baudrate=10000", but this should not be related to your problem I guess. Context [here](https://www.hackster.io/dragos-iosub/basic-iot-raspberrypi-ccs811-air-quality-monitoring-7b22e0). Anyway you could try to reproduce the problem by changing path and adding baudrate to config just to reproduce... 

Beside that I have no idea. Cool with your mushroom farm, I'm still dialing it in, hope to post some pictures soon :).

fabi2803, 2021-04-02:
Still same...

> pi@raspberrypi:~ $ python /usr/local/sbin/bh1750.py
> 87.50

> 2021-04-02 08:09:22,119 - WARNING - terrariumSensor      - Measured light value Nonelux from script sensor 'Licht' is outside valid range 0.00lux - 100.00lux in 0.06640 seconds.
> 2021-04-02 08:09:22,051 - INFO    - terrariumSensor      - Loaded script light sensor 'Licht' on location /usr/local/sbin/bh1750.py.

wolli1234, 2021-04-02:
Strange, no idea... here are my settings

[image]

theyosh, 2021-04-02:
Hmm, maybe I found something....

All your tests you run: `python /usr/local/sbin/bh1750.py` which is Python 2. What happens if you run `python3 /usr/local/sbin/bh1750.py`? This will run it with Python3.

I guess that running with Python3 will give an error? Because the script '/usr/local/sbin/bh1750.py' has the shebang to Python3 (first line: #!/usr/bin/python3)

If it give an error with Python3, you can change the first line of your script to: `#!/usr/bin/python2` and then it should work again (I hope)

fabi2803, 2021-04-02:
> edit: You should always run the script as follows: /usr/local/sbin/bh1750.py. If that give an error, try: sudo chmod +x /usr/local/sbin/bh1750.py and try again. The script needs to be executable.

Nice that was the problem! Thank you very much.