theyosh/TerrariumPI #530: [FEATURE]:Support for SEQUENT 4relay stackable board

morrell7, 2021-03-23:
**Is your feature request related to a problem? Please describe.**
I'm new to coding and would like to understand how to make sure the 4relay board will be compatible with TerrariumPi

**Describe the solution you'd like**
Simple guide or help to integrate this the same as 8 relay board.

**Additional context**
Worried 8 relay board won't carry the amps I need it to.

theyosh, 2021-03-25:
Hmm, I have added it to the new version. I am almost done with the 4.X version, and there I have added the 4 relay board.

In the current version it is more work to add it. So I am not adding it to the current 3.X version. Sorry.

morrell7, 2021-03-25:
I will happily wait :) Thank you for sharing your work

closed 2021-03-25 by morrell7