theyosh/TerrariumPI #531: [BUG]: Can't update login credentials using RaspberryPi Zero. (No sound card selected)

Sagi313, 2021-03-24:
Using Python3, and Raspberry Pi Zero W+ ( I know it's not supported, But it works!)

When i go into the setting menu, and want to change the default credentials from "admin/password", I get requested to select a sound card to continue. The problem is I don't have any sound card.

[image]

theyosh, 2021-03-25:
If you do a git pull, you should be able to select 'None' for soundcard. Make sure your browser cache is cleared.

closed 2021-03-25 by theyosh

Sagi313, 2021-03-26:
Thank you!
by the way- I really enjoy TerrariumPi

theyosh, 2021-03-26:
Nice! Did you post some pictures at #210 ?

Sagi313, 2021-03-27:
Not yet.
Going to post some as soon as i'll finish my makeover!

PR #532 (2021-03-25): Allow no soundcard as an option. #531
Should fix #531