theyosh/TerrariumPI #533: CCS811 refine readings with temperature and humidtity values?

wolli1234, 2021-03-25:
Hello,

I use the BME280 for temperature and humidity and the CCS811 for CO2. When humidity and temperature are known by the CCS811, it is able to refine it's tVOC and CO2 readings.

Is this taken into account in the current code?

`def setCompensation(self, temperature, humidity):
                temperature = round(temperature,2)
                humidity = round(humidity,2)
                print('Setting compensation to {} C and {} %'.format(temperature, humidity))
                hum1 = int(humidity//0.5)
                hum2 = int(humidity*512-hum1*256)

                temperature = temperature+25
                temp1 = int(temperature//0.5)
                temp2 = int(temperature*512-temp1*256)

                s = [CSS811_ENV_DATA,hum1,hum2,temp1,temp2,0x00]
                s2 = bytearray( s )
                CCS811_fw.write( s2 )
                
                return`

Or is this something else?

Greetings,
Wolli

_Originally posted by @wolli1234 in https://github.com/theyosh/TerrariumPI/issues/462#issuecomment-806531473_

theyosh, 2021-03-27:
Hi,

no, at the moment, the temperature and humidity is NOT taken into account. This is because then you have to select the sensors you want to use for the temperature and humidity. That is a bit hard at the moment.

But I think I can add this as a calibration option in my new version.

wolli1234, 2021-03-28:
Hey,

that would be great!

theyosh, 2021-03-29:
As the new version support multi environments and multi area's, it is not that simple just to pick a temperature and humidity sensor. So that is why I made the option to add extra compensation sensors in the 4.x version.

An alpha/beta version is just around the corner for testing...