theyosh/TerrariumPI #535: Custom Script Switch - not sure how this works

wolli1234, 2021-03-30:
Hello,

I'm trying to control a RGB LED strip as a switch in TerrariumPi. RGB-Color should change depending on different mushroom strains, which I control with software pwm from 0 to 255 (0% -> 100%). I wrote a little python script that should be enough to test, but it is not working within TerarriumPI. Beside I was using the external_switch.py example from the contrib folder, but as I'm not a developer I probably miss out something. When I run this adapted script in a terminal I get this error:

`Traceback (most recent call last):
  File "test_wolli2.py", line 20, in <module>
    raise SyntaxError('No arguments specified')
SyntaxError: No arguments specified`

Here is the full code I'm trying to run as a script within TerrariumPi:

`#!/usr/bin/env python
import sys
import pigpio

RED_PIN   = 17
GREEN_PIN = 22
BLUE_PIN  = 24

pi =pigpio.pi()

bright = 255

def setLights(pin, brightness):
	realBrightness = int(int(brightness) * (float(bright) / 255.0))
	pi.set_PWM_dutycycle(pin, realBrightness)

# This is just an example script to test the functionality for TerrariumPI. It is also a reference for implementing your own scripts

if len(sys.argv) == 1:
  raise SyntaxError('No arguments specified')

try:
  value = int(sys.argv[1])
except Exception as ex:
  raise ValueError('Invalid input value. Should be a number between 0 and 100 including')

if not (0 <= value <= 100):
  raise ValueError('Invalid input value. Should be a number between 0 and 100 including')

if value == 0:
  print('Toggle power switch to state OFF')
  # Call your code here....
  setLights(RED_PIN, 0)
  setLights(GREEN_PIN, 0)
  setLights(BLUE_PIN, 0)
else:
  print('Toggle power switch to state ON or dimming value: {}'.format(value))
  # Call your code here....
  setLights(RED_PIN, 255)
  setLights(GREEN_PIN, 0)
  setLights(BLUE_PIN, 128)`

Did I adapt the example wrong? Sorry, I should learn some python I guess :).

Best regards,
Wolli

theyosh, 2021-04-02:
If you want to test this script, you need to add a value to the command like: `script.py 0` or `script.py 50` or `script.py 100`.
The value is the amount of power it should give. With a 0 (zero) the dimmer is dimming at 100% (no light). with 100 the light is full on.

Make sure your script is executable: `cmod +x [where_your_script_is]` and run it like: `/[full/path/to/script] [dimmer_value]`

wolli1234, 2021-04-02:
> Make sure your script is executable: `chmod +x [where_your_script_is]` and run it like: `/[full/path/to/script] [dimmer_value]`

Thank you, chmod +x did the trick, now it's working!

theyosh, 2021-04-02:
Nice... I will add a check for this in the new version....

Did I asked you already for #210 ?

wolli1234, 2021-04-02:
> Did I asked you already for #210 ?

I will definitely post some pictures and description when the first mushrooms move in :). Still setting up the basics right now.

closed 2021-04-03 by wolli1234