theyosh/TerrariumPI #536: On several raspberry Pi 4 models having issues with gevent modules missing after a reboot.

Specialmias, 2021-04-01:
Hi there,
I have several RPI4b models with 8gbs of ram I decided to use with terrariumPI that all eventually had an issue with gevent missing after a reboot. The procedure for installing was as follows
1) Flash SD card with the official Pi OS flashing software. Selected the lite version of Raspberry pi OS. On boot I changed the default password and enabled SSH and the camera with raspi-config
2) Installed using the script sudo ./install.sh 3 
3) First boot worked fine but changing a setting to an invalid value (such as a weather url from weather underground) and doing a reboot would lead to one of two errors:
Import Error: no module named gevent or gevent library: Import Error: no module named oled.device
4) I noticed this would persist even with full reinstallations on different pi's (one wired in one room, one wireless). However, I noticed all of the errors were coming from python 2.7 I specifically installed via sudo ./install.sh 3
5) I was able to fix this by ensuring version 3.7 of Python was always the default with the command:

sudo update-alternatives --install /usr/bin/python python /usr/bin/python2.7 1
sudo update-alternatives --install /usr/bin/python python /usr/bin/python3.7 2

Now it works even with twister OS full running in the background. Weather stops the server from booting still even with values such as: 

https://api.openweathermap.org/data/2.5/weather?q=washington,DC,US&units=metric&appid=myapikeywashere

But otherwise I am completely stable now. I hope this is helpful someone out there. 
Thanks for the work on this dream software package. I think most exotic pet people have been cobbling together versions of this piecemeal and we've never had a whole kit like this.

theyosh, 2021-04-02:
Could you try to debug the error: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

It is strange that the weather is causing problems....

Specialmias, 2021-04-02:
Hi there,
turned on the debugging following the document and nothing obvious showed up there:
Apr  2 08:46:45 raspberrypi 2021-04-02 08:46:45,799 - INFO    - terrariumPI          - Starting terrariumPI
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,053 - INFO    - terrariumEngine      - Loading terrariumPI config
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,064 - INFO    - terrariumEngine      - Done Loading terrariumPI config
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,219 - INFO    - terrariumEngine      - Setting terrariumPI authentication
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,219 - INFO    - terrariumEngine      - Done setting terrariumPI authentication
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,220 - INFO    - terrariumEngine      - Loading terrariumPI collector
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,221 - INFO    - terrariumCollector   - Setting up collector database history.db
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,223 - INFO    - terrariumCollector   - Database connection created to database history.db
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,224 - INFO    - terrariumCollector   - Collector database is up to date
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,225 - INFO    - terrariumCollector   - TerrariumPI Collecter is ready
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,226 - INFO    - terrariumEngine      - Done loading terrariumPI collector
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,226 - INFO    - terrariumEngine      - Loading terrariumPI PI power setting
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,227 - INFO    - terrariumEngine      - Done loading terrariumPI PI power setting
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,227 - INFO    - terrariumEngine      - Loading terrariumPI PI temperature indicator
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,228 - INFO    - terrariumEngine      - Done loading terrariumPI PI temperature indicator
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,229 - INFO    - terrariumEngine      - Loading terrariumPI PI distance indicator
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,229 - INFO    - terrariumEngine      - Done loading terrariumPI PI distance indicator
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,230 - INFO    - terrariumEngine      - Loading terrariumPI PI windspeed indicator
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,231 - INFO    - terrariumEngine      - Done loading terrariumPI PI windspeed indicator
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,231 - INFO    - terrariumEngine      - Loading terrariumPI PI volume indicator
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,232 - INFO    - terrariumEngine      - Done loading terrariumPI PI volume indicator
Apr  2 08:46:46 raspberrypi 2021-04-02 08:46:46,232 - INFO    - terrariumEngine      - Loading terrariumPI weather data
Apr  2 08:46:50 raspberrypi 2021-04-02 08:46:50,757 - INFO    - terrariumPI          - Starting terrariumPI
Apr  2 08:46:51 raspberrypi 2021-04-02 08:46:51,023 - INFO    - terrariumEngine      - Loading terrariumPI config
Apr  2 08:46:51 raspberrypi 2021-04-02 08:46:51,033 - INFO    - terrariumEngine      - Done Loading terrariumPI config

However,  screen -r after starting the start.sh script yielded:
2021-04-02 08:53:47,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.86.70:8090 ...
2021-04-02 08:53:52,059 - WARNING - terrariumEngine      - Unable to get the latest version information from Github. Will check next round.
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 30, in <module>
    terrariumEngine = terrariumEngine()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 142, in __init__
    self.get_weather)
  File "/home/pi/TerrariumPI/terrariumWeather.py", line 431, in __new__
    raise terrariumWeatherSourceException()
terrariumWeather.terrariumWeatherSourceException
2021-04-02 08:53:53,000 - INFO    - terrariumWrapper     - Restart counter: 1/5
2021-04-02 08:53:53,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 6 seconds. Press Ctrl+C now to terminate TerrariumPI.

closed 2021-04-02 by Specialmias