theyosh/TerrariumPI #538: Snake wnclosure setup

tgrantjr, 2021-04-02:
I am wanting to set this up for 2 snakes. I want to have 5v rgb led for sunrise and sunset, 1 dht11 per enclosure as well as a humidifier on each. Can this software control this all with sensors and relays?  I can wire it up no problem. Just want to figure it out first. Thanks

cyber-shifter, 2021-04-02:
I, too, am using this for snakes. I have three enclosures that I will be using this for. I wouldn't use the DHT's personally. They're not a very reliable sensor. They're cheap, but not very good. Plus, you're going to be putting them in an environment that's humid, which they don't like very much. I would go with one of the other sensor options personally. I have radiant heat panels for my heat source in mine, so I use IR thermometers, with an enclosure around them. I use a dimmer pack to control the power to the heat panels, and a relay board to switch on some LED's to simulate day/night. It can all be done pretty easily. I would decide how you're going to heat, and then go from there. Supposedly the 4.0 version will include support for multiple environments in one instance, but I'm going to continue using multiple Pi's. Always buy extra parts, if one breaks suddenly, you don't want your animals to wait for the replacement.

theyosh, 2021-04-09:
@tgrantjr The current version (3.9.9) is not capable of controlling multiple enclosures. That is in the new version which is now open for testing. It is not fully finished yet. So that means either running with 2 PIs or wait a bit longer until my new version 4 is stable.

About using RGB leds I do not know. As far as I know, people use them with extra scripts. I do not support them directly

And for DHT, please switch to BME280. Way more reliable.

tgrantjr, 2021-04-09:
I'll jave to look into those sensors and figure out how to connect them. I was hoping until the new version comes out that I could assign a sensor to relays so when temp and humidity is out of spec they trip. That would accommodate 2 enclosures. I won't have it done for at least another month anyway so maybe 4 will be out by then.

theyosh, 2021-04-10:
Version 4 will support this. Currently, version 3 does not.

theyosh, 2021-09-01:
Hi, have you already tried version 4? It is released now. And I would like to close this issue. So is it still relevant?