theyosh/TerrariumPI #542: [BUG]: Error 500

GreeNNut42, 2021-04-13:
**Setup:**
 - Python version: 3.7.3
 - Raspberry PI: 2B

**Describe the bug**
 I installed  V3.9.9 following your tutorial, I launch the server ( secured server on web), and the I got Error 500, even in local way, the description is here : 
Internal Server Error

Exception:

OSError(24, 'Too many open files')

Traceback:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/bottle.py", line 868, in _handle
  File "/usr/local/lib/python3.7/dist-packages/bottle.py", line 1748, in wrapper
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 43, in webserver_headers
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 85, in wrapper
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 239, in __render_page
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 228, in __template_variables
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 1351, in get_profile_name
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 1342, in get_profile_config
OSError: [Errno 24] Too many open files: 'description.txt'

theyosh, 2021-04-14:
Hmm, this sounds like an OS issue. Too many open files means that something is not working correctly.

In order to debug look here for instructions: https://github.com/theyosh/TerrariumPI/wiki/FAQ#screen-3

Also you could try: `lsof -u pi` This will give a list of all open files by the user PI. Save this to a file and upload it here? Then I can have a look at it.

theyosh, 2021-07-26:
Closing due to no response and problem outside TerrariumPI

closed 2021-07-26 by theyosh