theyosh/TerrariumPI #543: [FEATURE]: New option to the Authentication- allow to turn switches on/off

Sagi313, 2021-04-13:
I turn my lights on/off manually in the "Switches" tab, but every time I wish to do so, I need to enter username and password.

I would be glad is there will be an option (beside "full authentication" and "changes only"), that allows to manage all the switches without authentication.

[image]

Maybe you can consider an option with no authentication at all 🔢 
Thank you

theyosh, 2021-04-14:
Hi,

you mean that you toggle the relay in the interface? It should remember the login until your close your browser.

I am not a fan of totally disabling authentication. As this can introduce security issues. Especially when people put it online to the internet.  And only making the relay changes without authentication is not an option.

So I could add a 'no authentication' option to it. But I am afraid that this will be done in the new version 4. As I will not make new features in the 3 version. We are already testing version 4.

Sagi313, 2021-04-15:
> you mean that you toggle the relay in the interface? It should remember the login until your close your browser.
Yes. I close the browser very often, so I have to do a full authentication each time.

I get your point about the security issues, that makes sense, although mine is only on the local network.

anyway, i'll be glad to see this feature in version 4!
Also- Going to upload a picture of my setup this week :)

theyosh, 2021-08-09:
Hi,

This is now possible in Version 4. If you have a spare Pi and some spare hardware, you could test this...