theyosh/TerrariumPI #544: [BUG]: Version 4 Advise....

ewiandr73, 2021-04-20:
This for version 4:

I have installed version 4 on a Rpi4 (also tried a 3B), but the Pi seems to keep resetting.
I can't see any issue and the pi is stable before I install the Terrarium SW.

I know this version is not supported yet, but I was wondering if you have seen this when you have been testing yours and if so what the issue was...

Thanks 

Andy

**Setup:**
 - Python version: [3]
 - Raspberry PI: [4]

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce / Installation steps**
Install as per the instruction, I use the lite version of pi OS.

**Expected behavior**
Stop resetting

theyosh, 2021-04-20:
Hi, yeah, this is due to the terrariumpi service startup... I have enabled watchdog functionality which will restart the Pi after 3 restarts in 10 minutes. This I have done, due to the fact my production systems are hanging, and I would like to automatically recover from that. 
But I have also seen that manually restarting will also trigger the watchdog. So I think I will remove it during the testing phase.

For now, you can fix it by putting the SD card into your pc and remove the file: /[sdcard]/etc/systemd/system/terrariumpi.service

Then you can restart your Pi and it should boot. But my software is not running because the startup script is not available.

theyosh, 2021-04-20:
Ok, I made an update of the code. So if you have your Pi running again, do a `git pull` and rerun the installer. This will remove the watchdog functionality for now.

And then try to start it, and look in the logs to see what the errors are. Because the restarts are triggered probably due to an error.

theyosh, 2021-04-22:
Did it solve the issue of rebooting? Then we can close this issue

ewiandr73, 2021-04-23:
I tried to rerun the installer but the time between restarts was too short to complete the install.

So I reinstalled everything from scratch and the RPI restart have stopped, but the Terrarium GUI seems to crash after a short time and doesn't seem to recover.  I check the logs, the 'terrariumpi.debug.log' is empty and the 'terrariumpi.log' just seems to repeat the following:

2021-04-23 10:07:01,504 - INFO    - terrariumEngine      - Starting a new update round with 0 sensors, 0 relays, 0 buttons and 0 webcams.
2021-04-23 10:09:09,803 - INFO    - terrariumEngine      - Starting TerrariumPI 4.0.0 ...
2021-04-23 10:09:09,862 - INFO    - terrariumEngine      - Loaded 27 settings in 0.06 seconds.
2021-04-23 10:09:10,577 - INFO    - terrariumEngine      - Loading existing sensors from database.
2021-04-23 10:09:10,582 - INFO    - terrariumEngine      - Scanning for new sensors ...
2021-04-23 10:09:10,684 - INFO    - merros_sensor        - Meross cloud is not enabled.
2021-04-23 10:09:16,799 - INFO    - terrariumEngine      - Loaded 0 sensors in 6.22 seconds.
2021-04-23 10:09:16,800 - INFO    - terrariumEngine      - Loading existing relays from database.
2021-04-23 10:09:16,811 - INFO    - terrariumEngine      - Scanning for new relays ...
2021-04-23 10:09:22,150 - INFO    - meross_relay         - Meross cloud is not enabled.
2021-04-23 10:09:32,178 - INFO    - terrariumEngine      - Loaded 0 relays in 15.38 seconds.
2021-04-23 10:09:32,179 - INFO    - terrariumEngine      - Loading existing buttons from database.
2021-04-23 10:09:32,188 - INFO    - terrariumEngine      - Loaded 0 buttons in 0.01 seconds.
2021-04-23 10:09:32,189 - INFO    - terrariumEngine      - Loading existing webcams from database.
2021-04-23 10:09:32,195 - INFO    - terrariumEngine      - Loaded 0 webcams in 0.01 seconds.
2021-04-23 10:09:32,196 - INFO    - terrariumEngine      - Loading existing enclosures from database.
2021-04-23 10:09:32,202 - INFO    - terrariumEngine      - Loaded 0 enclosures in 0.01 seconds.
2021-04-23 10:09:32,203 - INFO    - terrariumEngine      - TerrariumPI is up and running at address: http://SnakeMonitor:8090 in 23.05 seconds.
2021-04-23 10:09:32,204 - INFO    - terrariumEngine      - Starting log tailing.
2021-04-23 10:09:32,226 - INFO    - terrariumEngine      - Starting engine updater with 30.00 seconds interval.
2021-04-23 10:09:32,229 - INFO    - terrariumWebserver   - Running webserver at SnakeMonitor:8090
2021-04-23 10:09:32,480 - INFO    - terrariumEngine      - Starting a new update round with 0 sensors, 0 relays, 0 buttons and 0 webcams.

Are you seeing similar behaviours?

Regards

Andy

ewiandr73, 2021-04-23:
Also, I reboot the pi and the GUI still doesn't come back up.

theyosh, 2021-04-23:
The above log file looks right. So you should be able to access the interface at http://SnakeMonitor:8090

If that is repeating, then there has to be an error somewhere in the logging. I also do not see a logline with an error. So can't tell what is going on right now.

It can take some time or you have to refresh your browser a few times to get the web page. Then I think the Wifi signal is a bit low.

And, there is also some logging in /var/log/syslog. There you can look for lines like

```
python[11102]: [2021-04-23 13:39:03.382009] Start updating ....
python[11102]: [2021-04-23 13:39:12.019864] Done in 8.64 seconds. Waiting for 21.36 seconds for new round +/- 0.00 seconds.
```

This indicates that it is working and running. But there could be errors, so maybe you can find something there.

ewiandr73, 2021-04-23:
I still can't  login to the webpage using the IP or the hostname

This is from the /var/log/syslog

_Apr 24 09:16:17 raspberrypi kernel: [82822.053975] w1_master_driver w1_bus_master1: Attaching one wire slave 00.1c6000000000 crc 64
Apr 24 09:16:17 raspberrypi kernel: [82822.061751] w1_master_driver w1_bus_master1: Family 0 for 00.1c6000000000.64 is not registered.
Apr 24 09:16:33 raspberrypi systemd[1]: terrariumpi.service: Watchdog timeout (limit 2min)!
Apr 24 09:16:33 raspberrypi systemd[1]: terrariumpi.service: Killing process 13301 (python) with signal SIGABRT.
Apr 24 09:16:33 raspberrypi systemd[1]: terrariumpi.service: Killing process 13314 (tail) with signal SIGABRT.
Apr 24 09:16:33 raspberrypi systemd[1]: terrariumpi.service: Main process exited, code=killed, status=6/ABRT
Apr 24 09:16:33 raspberrypi systemd[1]: terrariumpi.service: Failed with result 'watchdog'.
Apr 24 09:16:38 raspberrypi systemd[1]: terrariumpi.service: Service RestartSec=5s expired, scheduling restart.
Apr 24 09:16:38 raspberrypi systemd[1]: terrariumpi.service: Scheduled restart job, restart counter is at 549.
Apr 24 09:16:38 raspberrypi systemd[1]: Stopped TerrariumPI.
Apr 24 09:16:38 raspberrypi systemd[1]: Starting TerrariumPI..._

theyosh, 2021-04-24:
Hmm, I need more logging. Can you look at the terrarium log file at `/home/pi/TerrariumPI/log/terrarium.log` around the same time?

It looks something is taking forever to finish.... And I guess in the terrariumpi logging you will find also lines saying that it is restarted. The just above that line, you can see what the last action was. Maybe there we can find what is causing this hanging..

The W1 errors should not be an issue. This indicates that there is not a 1wire connected to the  gpio pin.

ewiandr73, 2021-04-24:
This is the 2 logs for the same period

pi@raspberrypi:~/TerrariumPI/log$ tail -n 30 /var/log/syslog
Apr 24 19:15:22 raspberrypi python[1065]: 2021-04-24 19:15:22,973 - INFO    - terrariumEngine      - Loaded 0 enclosures in 0.01 seconds.
Apr 24 19:15:22 raspberrypi python[1065]: 2021-04-24 19:15:22,974 - INFO    - terrariumEngine      - TerrariumPI is up and running at address: http://SnakeMonitor:8090 in 23.05 seconds.
Apr 24 19:15:23 raspberrypi python[1065]: 2021-04-24 19:15:23,000 - INFO    - terrariumWebserver   - Running webserver at SnakeMonitor:8090
Apr 24 19:15:45 raspberrypi kernel: [ 3385.047585] w1_master_driver w1_bus_master1: Attaching one wire slave 00.820000000000 crc 30
Apr 24 19:15:45 raspberrypi kernel: [ 3385.056147] w1_master_driver w1_bus_master1: Family 0 for 00.820000000000.30 is not registered.
Apr 24 19:15:56 raspberrypi ntpd[583]: 129.250.35.251 local addr 192.168.86.26 -> <null>
Apr 24 19:16:50 raspberrypi kernel: [ 3450.008760] w1_master_driver w1_bus_master1: Attaching one wire slave 00.420000000000 crc fa
Apr 24 19:16:50 raspberrypi kernel: [ 3450.017576] w1_master_driver w1_bus_master1: Family 0 for 00.420000000000.fa is not registered.
Apr 24 19:17:01 raspberrypi CRON[1162]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Apr 24 19:17:23 raspberrypi systemd[1]: terrariumpi.service: Watchdog timeout (limit 2min)!
Apr 24 19:17:23 raspberrypi systemd[1]: terrariumpi.service: Killing process 1065 (python) with signal SIGABRT.
Apr 24 19:17:23 raspberrypi systemd[1]: terrariumpi.service: Killing process 1115 (tail) with signal SIGABRT.
Apr 24 19:17:23 raspberrypi systemd[1]: terrariumpi.service: Main process exited, code=killed, status=6/ABRT
Apr 24 19:17:23 raspberrypi systemd[1]: terrariumpi.service: Failed with result 'watchdog'.
Apr 24 19:17:28 raspberrypi systemd[1]: terrariumpi.service: Service RestartSec=5s expired, scheduling restart.
Apr 24 19:17:28 raspberrypi systemd[1]: terrariumpi.service: Scheduled restart job, restart counter is at 23.
Apr 24 19:17:28 raspberrypi systemd[1]: Stopped TerrariumPI.
Apr 24 19:17:28 raspberrypi systemd[1]: Starting TerrariumPI...
Apr 24 19:17:31 raspberrypi systemd[1]: Started TerrariumPI.
Apr 24 19:17:31 raspberrypi python[1179]: 2021-04-24 19:17:31,339 - INFO    - terrariumEngine      - Starting TerrariumPI 4.0.0 ...
Apr 24 19:17:31 raspberrypi python[1179]: 2021-04-24 19:17:31,398 - INFO    - terrariumEngine      - Loaded 27 settings in 0.06 seconds.
Apr 24 19:17:32 raspberrypi python[1179]: 2021-04-24 19:17:32,115 - INFO    - terrariumEngine      - Loading existing sensors from database.
Apr 24 19:17:32 raspberrypi python[1179]: 2021-04-24 19:17:32,119 - INFO    - terrariumEngine      - Scanning for new sensors ...
Apr 24 19:17:32 raspberrypi python[1179]: 2021-04-24 19:17:32,222 - INFO    - merros_sensor        - Meross cloud is not enabled.
Apr 24 19:17:38 raspberrypi python[1179]: 2021-04-24 19:17:38,353 - INFO    - terrariumEngine      - Loaded 0 sensors in 6.24 seconds.
Apr 24 19:17:38 raspberrypi python[1179]: 2021-04-24 19:17:38,354 - INFO    - terrariumEngine      - Loading existing relays from database.
Apr 24 19:17:38 raspberrypi python[1179]: 2021-04-24 19:17:38,365 - INFO    - terrariumEngine      - Scanning for new relays ...
Apr 24 19:17:38 raspberrypi python[1179]: No GEMBIRD SiS-PM found. Check USB connections, please!
Apr 24 19:17:42 raspberrypi kernel: [ 3501.969626] w1_master_driver w1_bus_master1: Attaching one wire slave 00.c20000000000 crc 76
Apr 24 19:17:42 raspberrypi kernel: [ 3501.978246] w1_master_driver w1_bus_master1: Family 0 for 00.c20000000000.76 is not registered.

pi@raspberrypi:~/TerrariumPI/log$ tail -n 30 terrariumpi.log
2021-04-24 19:15:00,586 - INFO    - terrariumEngine      - Starting TerrariumPI 4.0.0 ...
2021-04-24 19:15:00,650 - INFO    - terrariumEngine      - Loaded 27 settings in 0.06 seconds.
2021-04-24 19:15:01,366 - INFO    - terrariumEngine      - Loading existing sensors from database.
2021-04-24 19:15:01,370 - INFO    - terrariumEngine      - Scanning for new sensors ...
2021-04-24 19:15:01,473 - INFO    - merros_sensor        - Meross cloud is not enabled.
2021-04-24 19:15:07,591 - INFO    - terrariumEngine      - Loaded 0 sensors in 6.22 seconds.
2021-04-24 19:15:07,591 - INFO    - terrariumEngine      - Loading existing relays from database.
2021-04-24 19:15:07,599 - INFO    - terrariumEngine      - Scanning for new relays ...
2021-04-24 19:15:12,920 - INFO    - meross_relay         - Meross cloud is not enabled.
2021-04-24 19:15:22,949 - INFO    - terrariumEngine      - Loaded 0 relays in 15.36 seconds.
2021-04-24 19:15:22,950 - INFO    - terrariumEngine      - Loading existing buttons from database.
2021-04-24 19:15:22,959 - INFO    - terrariumEngine      - Loaded 0 buttons in 0.01 seconds.
2021-04-24 19:15:22,960 - INFO    - terrariumEngine      - Loading existing webcams from database.
2021-04-24 19:15:22,967 - INFO    - terrariumEngine      - Loaded 0 webcams in 0.01 seconds.
2021-04-24 19:15:22,967 - INFO    - terrariumEngine      - Loading existing enclosures from database.
2021-04-24 19:15:22,973 - INFO    - terrariumEngine      - Loaded 0 enclosures in 0.01 seconds.
2021-04-24 19:15:22,974 - INFO    - terrariumEngine      - TerrariumPI is up and running at address: http://SnakeMonitor:8090 in 23.05 seconds.
2021-04-24 19:15:22,975 - INFO    - terrariumEngine      - Starting log tailing.
[... cut ...]
2021-04-24 19:17:32,119 - INFO    - terrariumEngine      - Scanning for new sensors ...
2021-04-24 19:17:32,222 - INFO    - merros_sensor        - Meross cloud is not enabled.
2021-04-24 19:17:38,353 - INFO    - terrariumEngine      - Loaded 0 sensors in 6.24 seconds.
2021-04-24 19:17:38,354 - INFO    - terrariumEngine      - Loading existing relays from database.
2021-04-24 19:17:38,365 - INFO    - terrariumEngine      - Scanning for new relays ...
2021-04-24 19:17:43,701 - INFO    - meross_relay         - Meross cloud is not enabled.

theyosh, 2021-04-24:
Do you use 1-wire? If not, than you can change the following file: `/boot/config.txt` and find the line `dtoverlay=w1-gpio` and make sure there is a `#` in front of it.
Then reboot your PI. This will disable the 1-wire overlay, and I hope will start my software.

If you have 1-wire connected, then that is probably not correct. And have to look in the wiring of that.

ewiandr73, 2021-04-25:
I've done that and it looks like it's stable from the logs but I still can't access the web gui?

pi@raspberrypi:~/TerrariumPI/log$ tail -n 40 terrariumpi.log
2021-04-25 17:53:51,470 - INFO    - terrariumEngine      - Loading existing sensors from database.
2021-04-25 17:53:51,475 - INFO    - terrariumEngine      - Scanning for new sensors ...
2021-04-25 17:53:51,646 - INFO    - merros_sensor        - Meross cloud is not enabled.
2021-04-25 18:00:13,468 - INFO    - terrariumEngine      - Loaded 0 sensors in 382.00 seconds.
2021-04-25 18:00:13,469 - INFO    - terrariumEngine      - Loading existing relays from database.
2021-04-25 18:00:13,482 - INFO    - terrariumEngine      - Scanning for new relays ...
2021-04-25 18:00:19,597 - INFO    - meross_relay         - Meross cloud is not enabled.
2021-04-25 18:00:29,627 - INFO    - terrariumEngine      - Loaded 0 relays in 16.16 seconds.
2021-04-25 18:00:29,628 - INFO    - terrariumEngine      - Loading existing buttons from database.
2021-04-25 18:00:29,640 - INFO    - terrariumEngine      - Loaded 0 buttons in 0.01 seconds.
2021-04-25 18:00:29,641 - INFO    - terrariumEngine      - Loading existing webcams from database.
2021-04-25 18:00:29,647 - INFO    - terrariumEngine      - Loaded 0 webcams in 0.01 seconds.
2021-04-25 18:00:29,648 - INFO    - terrariumEngine      - Loading existing enclosures from database.
2021-04-25 18:00:29,657 - INFO    - terrariumEngine      - Loaded 0 enclosures in 0.01 seconds.
2021-04-25 18:00:29,658 - INFO    - terrariumEngine      - TerrariumPI is up and running at address: http://SnakeMonitor:8090 in 400.02 seconds.
2021-04-25 18:00:29,659 - INFO    - terrariumEngine      - Starting log tailing.
2021-04-25 18:00:29,683 - INFO    - terrariumEngine      - Starting engine updater with 30.00 seconds interval.
2021-04-25 18:00:29,686 - INFO    - terrariumWebserver   - Running webserver at SnakeMonitor:8090
2021-04-25 18:00:29,936 - INFO    - terrariumEngine      - Starting a new update round with 0 sensors, 0 relays, 0 buttons and 0 webcams.
2021-04-25 18:02:38,116 - INFO    - terrariumEngine      - Starting TerrariumPI 4.0.0 ...
2021-04-25 18:02:38,175 - INFO    - terrariumEngine      - Loaded 27 settings in 0.06 seconds.
2021-04-25 18:02:38,888 - INFO    - terrariumEngine      - Loading existing sensors from database.
2021-04-25 18:02:38,892 - INFO    - terrariumEngine      - Scanning for new sensors ...
2021-04-25 18:02:38,995 - INFO    - merros_sensor        - Meross cloud is not enabled.
2021-04-25 18:02:45,126 - INFO    - terrariumEngine      - Loaded 0 sensors in 6.24 seconds.
2021-04-25 18:02:45,127 - INFO    - terrariumEngine      - Loading existing relays from database.
2021-04-25 18:02:45,138 - INFO    - terrariumEngine      - Scanning for new relays ...
2021-04-25 18:02:51,005 - INFO    - meross_relay         - Meross cloud is not enabled.
2021-04-25 18:03:01,034 - INFO    - terrariumEngine      - Loaded 0 relays in 15.91 seconds.
2021-04-25 18:03:01,034 - INFO    - terrariumEngine      - Loading existing buttons from database.
2021-04-25 18:03:01,044 - INFO    - terrariumEngine      - Loaded 0 buttons in 0.01 seconds.
2021-04-25 18:03:01,044 - INFO    - terrariumEngine      - Loading existing webcams from database.
2021-04-25 18:03:01,051 - INFO    - terrariumEngine      - Loaded 0 webcams in 0.01 seconds.
2021-04-25 18:03:01,051 - INFO    - terrariumEngine      - Loading existing enclosures from database.
2021-04-25 18:03:01,057 - INFO    - terrariumEngine      - Loaded 0 enclosures in 0.01 seconds.
2021-04-25 18:03:01,058 - INFO    - terrariumEngine      - TerrariumPI is up and running at address: http://SnakeMonitor:8090 in 23.60 seconds.
2021-04-25 18:03:01,059 - INFO    - terrariumEngine      - Starting log tailing.
2021-04-25 18:03:01,085 - INFO    - terrariumEngine      - Starting engine updater with 30.00 seconds interval.
2021-04-25 18:03:01,089 - INFO    - terrariumWebserver   - Running webserver at SnakeMonitor:8090
2021-04-25 18:03:01,339 - INFO    - terrariumEngine      - Starting a new update round with 0 sensors, 0 relays, 0 buttons and 0 webcams.
pi@raspberrypi:~/TerrariumPI/log$ tail -n 40 /var/log/syslog
Apr 25 18:00:19 raspberrypi python[556]: 2021-04-25 18:00:19,597 - INFO    - meross_relay         - Meross cloud is not enabled.
Apr 25 18:00:26 raspberrypi systemd[1]: systemd-hostnamed.service: Succeeded.
Apr 25 18:00:29 raspberrypi python[556]: 2021-04-25 18:00:29,627 - INFO    - terrariumEngine      - Loaded 0 relays in 16.16 seconds.
Apr 25 18:00:29 raspberrypi python[556]: 2021-04-25 18:00:29,628 - INFO    - terrariumEngine      - Loading existing buttons from database.
Apr 25 18:00:29 raspberrypi python[556]: 2021-04-25 18:00:29,640 - INFO    - terrariumEngine      - Loaded 0 buttons in 0.01 seconds.
Apr 25 18:00:29 raspberrypi python[556]: 2021-04-25 18:00:29,641 - INFO    - terrariumEngine      - Loading existing webcams from database.
Apr 25 18:00:29 raspberrypi python[556]: 2021-04-25 18:00:29,647 - INFO    - terrariumEngine      - Loaded 0 webcams in 0.01 seconds.
Apr 25 18:00:29 raspberrypi python[556]: 2021-04-25 18:00:29,648 - INFO    - terrariumEngine      - Loading existing enclosures from database.
Apr 25 18:00:29 raspberrypi python[556]: 2021-04-25 18:00:29,657 - INFO    - terrariumEngine      - Loaded 0 enclosures in 0.01 seconds.
[... cut ...]
Apr 25 18:03:01 raspberrypi python[623]: 2021-04-25 18:03:01,051 - INFO    - terrariumEngine      - Loading existing enclosures from database.
Apr 25 18:03:01 raspberrypi python[623]: 2021-04-25 18:03:01,057 - INFO    - terrariumEngine      - Loaded 0 enclosures in 0.01 seconds.
Apr 25 18:03:01 raspberrypi python[623]: 2021-04-25 18:03:01,058 - INFO    - terrariumEngine      - TerrariumPI is up and running at address: http://SnakeMonitor:8090 in 23.60 seconds.
Apr 25 18:03:01 raspberrypi python[623]: 2021-04-25 18:03:01,000 - INFO    - terrariumWebserver   - Running webserver at SnakeMonitor:8090

ewiandr73, 2021-04-25:
Sorry not sure why the last post stuck out the text from the 1st log?

theyosh, 2021-04-25:
Hmm, it takes to much to time to start. I see values of 400 seconds. That is more than 5 minutes... That does not look right.

This could happen due to wrong time on the PI. I see it sometimes when NTP is a bit late in updating the clock.

But then, after 1 restart, NTP should done its work and it should start. So I have no clue at the moment why it is not working for you.

If you are handy with port forwarding and you trust me, you could setup a SSH port forward to your PI and I will have a look. If you are willing to do so, send login details to: github@theyosh.nl

theyosh, 2021-07-19:
Any updates? Does it work?

theyosh, 2021-07-26:
Closing due to no updates.... Try the latest code and see if the issue is still there.

closed 2021-07-26 by theyosh