theyosh/TerrariumPI #545: ESP8266 sensor data

se7en9057, 2021-04-20:
So i read this may be possible / unfeasible however i'm still curious. I have a thingsboard server running at work i use ESP8266 + sensors to send telemetry to (temp/humidity) so i know they will work in some cases. I was wondering if anyone has a working arduino sketch to send data telemetry to Terrarium PI would be handy to have sensors not daisy chained to my Pi or so that i can pull sensor data from other places not directly connected to my pi. 

Equipment refference WiMOS D1 Mini variant https://www.amazon.com/gp/product/B076F52NQD/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

theyosh, 2021-04-20:
Hi,

where does the data go now? That could maybe help.

Else I can only think of making a webserver on the device: https://www.allaboutcircuits.com/projects/using-an-arduino-as-a-web-server/
But I do not know if you can run multiple threads/processes. Because I would make in such a way, that there is a webserver process running, and a measuring process. So every X seconds a new measurement is done, and that is then written in a small JSON file. And that JSON file can then be loaded through the webserver.

That way, you can add it as a remote sensor to TerrariumPI. But I do not program in Arduino. So this is what I can think of.

theyosh, 2021-04-29:
Did the above suggestions help you? I think going with a WiFi + simple webserver should be the way to do it.

theyosh, 2021-07-19:
Closing due to no response

closed 2021-07-19 by theyosh