theyosh/TerrariumPI #546: [BUG]: Resiliency if connection to Bluetooth sensor is lost

duenni, 2021-04-23:
**Setup:**
 - Python version: 3
 - Raspberry PI: 3B

**Describe the bug**
I use TerrariumPi version 3.9.9 and besides other sensors also a Xiaomi Miflora Bluetooth sensor.
Every now and then TerrariumPi loses connection to the sensor. TerrariumEngine will then freak out after a while and can not keep up. I'll receive hundreds of mails within a few hours stating:
```Updating is having problems keeping up. Could not update in 30 seconds for 10 times!```
Fix is easy: reboot the Pi. The Miflora will reconnect. Seems like TerrariumEngine isn't trying to reconnect immediately when it loses the connection but only at startup.

**To Reproduce / Installation steps**
Connect a Miflora sensor to TerrariumPi. Wait for a disconnect or just pull out the battery.

**Expected behavior**
TerrariumPi should try to reconnect the bluetooth devices or restart the bluetooth stack. If that doesn't work a mail should be send out.
It would be enough to send out one email instead of an email every few minutes until the problem is fixed. TerrariumEngine should ignore the sensor if a certain amount of reconnection attempts is unsuccessful. This should be indicated in the GUI.

**Screenshots**
No screenshots but the log will put this out every few seconds:
```2021-04-15 13:55:38,624 - WARNING - terrariumBluetoothSensor - Error getting new data from miflora sensor 'miflora_light'. Error message: Failed to connect to peripheral xx:xx:xx:xx:xx:xx, addr type: public```

**Additional context**
It will take some time untile TerrariumEngine can not keep up anymore. Warnings add up in the log first.

theyosh, 2021-04-23:
Hi,

I have this also, and for me that is an indicator that the battery of the bluetooth device is getting low.

But I have fixed this in the new version 4. There I have a different approach on reading the sensors and reconnecting to them. Especially bluetooth sensors.
Because the first problem is that the bleutooth is hanging, and I did not have a timeout for reading new values. So that give a lot of issues. In version 4, I limited the readout to max 15 sec. After that I kill the readout.
Then I look if there are some bluepy-helper processes hanging, and try to kill them. 
If that all is not working, then there is a 2 minute timeout, which will trigger a restart of my software. So when it hangs to long, it will be restarted.
And finally, when 3 restarts in 10 minutes happens, it reboots the Pi totally. (This currently disabled during testing).

With this new logic, I do not have any hanging systems anymore. But there are still hangs and now restarts. So the system will recover better.

So, I am a bit sorry to say, but this will not be fixed in the 3.X version. As this is addressed (and hopefully) fixed in version 4.
A half solution could be to make a cron job, that restarts the bluetooth service every hour. I use that also, and I get the feeling that would also make the bluetooth sensors a bit more stable.

If you want you can test version 4: #541

duenni, 2021-04-24:
Hi

no problem at all. I'll wait for version 4 then, I guess. 😃 
Thanks for explaining.

theyosh, 2021-04-29:
Then I close it for now.

You can still make a cron job where it restarts the bluetooth service/stack every hour... Saves me some restarts. 

And it would really help if you could test version 4. #541

closed 2021-04-29 by theyosh