theyosh/TerrariumPI #547: Doubt with humidity and temperature sensor configuration

vikrilow, 2021-04-27:
Hello,

I'm configure my terrarium for tortoises. Now I have it:
* 1 HIB Lamp (this lamp provides temp and light, so, i need to use only during the day)
* 1 ceramic lamp (to use only at night if possible, but I think this isn't possible)
* 2 temp sensors ( one for each lamp)
* 2 humidity sensors ( one for each lamp)
* 3 mini fans configured to start 5 minutes each 4 hours (To renove air from terrarium)
* 2 fans configured to start if temp or humidity is hot.

So, I'd like to configure next Actions:

TEMP (Configuration for day and night, only with grades of diference, about 8º)
If it's Hot, HIB Lamp can't stop, because it gives hot and UVB, so I'll active fans to try cool air.
If it's Cold, Ceramic LAMP have to turn on. The problem with this configuration is that HIB and Ceramic lamp are in diferent sites(HIB is in the middle and ceramic is on one of the side) so that, How control temperature really?
I've thought next configuration:
Sensor temp HIB --> HOT --> Turn on fans to cool air
Sensor temp HIB --> COLD--> Nothing, tortoise will moves to hot zone under ceramic lamp

Sensor temp ceramic --> HOT --> Turn off ceramic lamp
Sensor temp ceramic --> COOL--> Turn on ceramic lamp

HUMIDITY:

There are 2 humidity sensors, so I think better option is work with media about both.

Sensor humidity ceramic --> HIGHT--> Turn on fans 
Sensor humidity ceramic --> LOW --> Turn on himidify machine

What do you think about this approach? It's correct? Any upgrading? Is this configuration possible in terrariumPi?

theyosh, 2021-04-30:
Hi,

at the moment, it is not possible to have two heating setups/areas. This is supported in version 4.

What you could do is just combine the temperature sensors and use for low alarm the ceramic lamp, and with high alarm the fan.
As your HIB lamp is basically running as daylight, you should setup that up with the lights environment part.

And then it is about tuning the temperature sensors. When you combine them, they take the averages of the alarm min and max values. So if you HIB sensor is around 30degrees, and you want to turn the ceramic lamp on when it gets at that location below 20degrees, you get the following formula:

20 + 30 = 50 degrees / 2 is 25 degrees is the minimal wanted average temperature. So that means that both low alarm values should average at 25. If you put both on 25 as a low alarm value, this should do the trick. The ceramic heater will turn on, when the temperature is getting below 20 degrees. Even that we have specified a higher low alarm for the the ceramic heater.

For the high alarm you do the same thing... tuning the sensor alarm values to get the required averages.

I hope it is clear what I try to explain. 

For humidity it is the same thing. Combine them, and tune the humidity sensors.

But, make sure that no fans are shared between environment parts. So the fan for humidity needs to be a different one than the cooler fan.

theyosh, 2021-07-19:
Is the above clear? Do you have an update?

theyosh, 2021-07-26:
Closing due to no response

closed 2021-07-26 by theyosh

vikrilow, 2021-07-26:
Hi
I'm sorry but I haven't just tried it, because I'm out of home.

Thanks

Saludos,
Víctor Lindo

Disculpe la brevedad y posibles faltas ortográficas, escrito desde
dispositivo móvil.