theyosh/TerrariumPI #548: Wifi and 4x20 display

tgrantjr, 2021-05-07:
I have everything setup. I need to have the pi connected to wifi and can't find the setting. I'm also wondering what I have to do to have my display connected and show current Temps and humidity. Is this possible?

martynwheeler, 2021-05-07:
I wrote a small python script to display the current temperature and run it
as a cron job every minute. I can dig it out if it's helpful?

theyosh, 2021-05-09:
@tgrantjr Wifi is just following: https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md

For the screen, I am afraid that you have to make something for your self as @martynwheeler did. I do show some summary on the LCD display, but with 4 x 20 display, it is hard to make it useful.

So, I advise always to make something for yourself. As then you can make it like you want it.

tgrantjr, 2021-05-09:
I have that figured out. I just can't get the wifi to work. It's connected
and I can ping outside but I can't get to the website to manage it

tgrantjr, 2021-05-10:
I did the install with python 3 as you instructed. I have it connected to my TV for testing and it does not come up with the daily message. I tried stopping and starting it and it's an invalid command. I'm going to try again with a fresh install.

tgrantjr, 2021-05-17:
It was not starting with the pi. I had to run the start.sh in the directory to get it to work

theyosh, 2021-07-19:
Hmm, is it working now? As it is not clear to me what the problem is. Wifi connection is outside my installation scope. So that should be working already before installing my software.

Can you give a status update

closed 2021-08-23 by tgrantjr