theyosh/TerrariumPI #553: [FEATURE]: Using IR thermometer sensor to regulate basking surface temperature

killwater, 2021-05-30:
I have an issue with IR thermometer readout of the basking spot - when the lizard will go there sensor will measure avarage temperature of the rock and the lizard together dropping down the value and triggering the basking lamp switch even if not needed (rock being very hot already). Does anyone have any experience with that?
The only solution I can see is to set the basking spot to ambient temperature and once lizard comes on the spot and temperature drops extra basking bulb is trigerred. This would save some electricity as well as it would only work when the basking is in progress.

theyosh, 2021-06-16:
Hi,

Maybe a totally different approach: Is it possible to drill a small hole in the rock and put a DS18B20 sensor in it? https://www.espruino.com/DS18B20
Then you can measure the inside of the rock, and when the lizard will climb on the rock, the temperature will not drop that fast and low.

Or else, with TP3 there is not a real good solution. With TP4 you can set an amount of errors to wait, before trigger on the relay. But this is also the case when the temperature is just getting to low, and needs to heat up. That will then also have a delay.

But, I would go for a DS18B20 sensor and try to drill a small hole in the rock itself....

theyosh, 2021-07-19:
Hi, 
I do not have a different solution than trying to put a sensor inside the rock and use that for measurement.

So can we close this issue?

killwater, 2021-07-19:
Hello
I am just finishing a new enclousure. Once I get everything in place I will run a couple of tests with the IR thermometer. If there will be something that can be resolved in the software can I open a new issue?

theyosh, 2021-07-19:
Sure, no problem. But the 'issue' will still be there. I can't see why the temperature is dropping. Therefore, you have to act if it is getting to cold.

So, a smaller light could maybe a solution for that.

But than we can leave it open for now. Waiting on your results. No problem.

theyosh, 2021-08-09:
I will close it now. Open it when you have updates or new questions... For now, I still suggesting for a DS1820 sensor in a rock..

closed 2021-08-09 by theyosh