theyosh/TerrariumPI #556: [BUG]:

Plantbrowser, 2021-06-18:
**Setup:**
 - Python Version: 2.7.16
 - Raspberry PI 3 B+

**Describe the bug**
During the day, the power switch for cooling is turn on when the terrarium goes past 75F. It will turn off the power switch after it goes below 75F. During the night, the power switch does not turn on when the terrarium temperature is past the new max temperature of 60F. No cooling no temperature drop I need for my terrarium. 

**To Reproduce / Installation steps**
Please provide the steps to reproduce or how you have installed the software.

**Expected behavior**
During the day, the power switch for cooling will turn on for 15 minutes if the terrarium goes past 75F. It will turn off the power switch for cooling when the terrarium temperature is below 75F. During the night, the night difference is -15F. The new temperature max is 60F. The power switch will turn on in 15 minute intervals until the terrarium temperature is below 60F. 

**Screenshots**
[image]
[image]
[image]
[image]
[image]

I know the temperature is below 65F. I had to reboot several times to see if I could get it to work somehow. The new temperature max is -60F and the power switch for the chiller and pump does not turn on still. I'm not sure what to do. 

**Additional context**
- From screen -r
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
2021-06-17 22:53:38,002 - WARNING - terrariumSensor      - Measured temperature value 64.37241586476566F from bme280 sensor 'Ambient Temperature' is erratic compared to previous value 17.97440923547838F in 0.23414 seconds.
2021-06-17 22:53:38,349 - WARNING - terrariumSensor      - Measured humidity value 79.7090663230742% from bme280 sensor 'Humidity' is erratic compared to previous value 79.83076433534767% in 0.23317 seconds.
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
2021-06-17 22:54:08,081 - WARNING - terrariumSensor      - Measured temperature value 64.38165548600372F from bme280 sensor 'Ambient Temperature' is erratic compared to previous value 17.97440923547838F in 0.23392 seconds.
2021-06-17 22:54:08,429 - WARNING - terrariumSensor      - Measured humidity value 79.82628343089586% from bme280 sensor 'Humidity' is erratic compared to previous value 79.83076433534767% in 0.23318 seconds.
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
2021-06-17 22:54:38,081 - WARNING - terrariumSensor      - Measured temperature value 64.39089510776569F from bme280 sensor 'Ambient Temperature' is erratic compared to previous value 17.97440923547838F in 0.23395 seconds.
2021-06-17 22:54:38,429 - WARNING - terrariumSensor      - Measured humidity value 79.89679427983694% from bme280 sensor 'Humidity' is erratic compared to previous value 79.83076433534767% in 0.23313 seconds.
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
2021-06-17 22:55:08,166 - WARNING - terrariumSensor      - Measured temperature value 64.44633284933866F from bme280 sensor 'Ambient Temperature' is erratic compared to previous value 17.97440923547838F in 0.23338 seconds.
2021-06-17 22:55:08,518 - WARNING - terrariumSensor      - Measured humidity value 79.86402877503066% from bme280 sensor 'Humidity' is erratic compared to previous value 79.83076433534767% in 0.23317 seconds.
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
2021-06-17 22:55:38,161 - WARNING - terrariumSensor      - Measured temperature value 64.44633284933866F from bme280 sensor 'Ambient Temperature' is erratic compared to previous value 17.97440923547838F in 0.23324 seconds.
2021-06-17 22:55:38,163 - WARNING - terrariumSensor      - After 5 erratic temperature measurements is the current value 64.44633284933866F is promoted to a valid value for bme280 sensor 'Ambient Temperature' in 0.23454 seconds.
2021-06-17 22:55:38,287 - WARNING - terrariumSensor      - Measured humidity value 79.80609481490659% from bme280 sensor 'Humidity' is erratic compared to previous value 79.83076433534767% in 0.00023 seconds.
2021-06-17 22:55:38,289 - WARNING - terrariumSensor      - After 5 erratic humidity measurements is the current value 79.80609481490659% is promoted to a valid value for bme280 sensor 'Humidity' in 0.00161 seconds.
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
2021-06-17 22:56:08,161 - WARNING - terrariumSensor      - Measured temperature value 64.44633284933866F from bme280 sensor 'Ambient Temperature' is erratic compared to previous value 18.025740471854807F in 0.23330 seconds.
2021-06-17 22:56:08,509 - WARNING - terrariumSensor      - Measured humidity value 79.8308063560534% from bme280 sensor 'Humidity' is erratic compared to previous value 79.80609481490659% in 0.23317 seconds.
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
2021-06-17 22:56:38,241 - WARNING - terrariumSensor      - Measured temperature value 64.4648121007206F from bme280 sensor 'Ambient Temperature' is erratic compared to previous value 18.025740471854807F in 0.23323 seconds.
2021-06-17 22:56:38,589 - WARNING - terrariumSensor      - Measured humidity value 79.82541963958309% from bme280 sensor 'Humidity' is erratic compared to previous value 79.80609481490659% in 0.23328 seconds.
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED
Current client connection status is: ClientStatus.SUBSCRIBED

bggangel, 2021-06-18:
I use temp low/high and it works fine. I did notice the following though.... 

"After 5 erratic temperature measurements is the current value 64.44633284933866F is promoted to a valid value for bme280 sensor 'Ambient Temperature' in 0.23454 seconds."

Then the next time reading for Ambient Temperature is:
2021-06-17 22:56:08,161 - WARNING - terrariumSensor - Measured temperature value 64.44633284933866F from bme280 sensor 'Ambient Temperature' is erratic compared to previous value 18.025740471854807F in 0.23330 seconds.

I do not believe in coincidence. 64.44633 Fahrenheit is 18.0257388 Celsius.
So for some reason, the "previous value" is stored or read as Celcius and compared to the Fahrenheit value.

I am not sure if this is something you (Plantbrowser) has done or if something is wrong with the software, but as stated above, I use a 100% Fahrenheit setup and it works fine for me, so there is probably something wrong with your Temperature Unit Setup

Plantbrowser, 2021-06-18:
How do I make sure my setup is 100%? I thought it was 100% when you switch it to Fahrenheit in account settings.

bggangel, 2021-06-19:
Don't know that, just figured that I shared what I saw. I have not changed units (nor do I know how to change them).

theyosh, 2021-07-19:
Hi,

sorry for the long delay. Is this still an issue? Because I also see negative numbers on the screenshots which does not look correct.

There is one issue with this, and that is that you should not change the sensors settings during your 'night time' period. Long story, but it does not work.

So, setup the sensors during the day, the way you want them. And then setup the day/night temperature values. And let it running. It looked ok about the setup. I think rebooting and maybe updating the sensors during the 'night period' is causing this.

Also, when you switch from Celsius to Fahrenheit, only the current values of the sensors are converted. Not the alarm and min/max values. The only setting is in the settings page.

closed 2021-07-19 by Plantbrowser