theyosh/TerrariumPI #557: Virtualbox virtualbox raspberry pi installation

LuisMoraes132, 2021-06-22:
Hey how is everything.  I would like to know if this system works in VirtualBox, simulating a breath for me, I did the installation correctly, but it's not opening the external IP through VirtualBox, can you tell me or if there is a video teaching how to install.

LuisMoraes132, 2021-06-22:
[image]

LuisMoraes132, 2021-06-22:
[image]

theyosh, 2021-06-22:
Hi,

sorry to say, but VirtualBox is not supported. Therefore I cannot support this. You are on you own with this.

you can try to stat manully to see what is happening:

`cd /home/pi/TerrariumPI`
`python3 terrariumPI.py`

And see what happens... But I am afraid that this is where the support stops...

theyosh, 2021-07-19:
Closing as this is not supported.

closed 2021-07-19 by theyosh