theyosh/TerrariumPI #559: [FEATURE]: support Energenie Pimote TWO WAY (ENER314-RT)

rossisbudda, 2021-07-01:
Hello. Great, great project! Please excuse any noob-ness, I'm learning!!

I see that we support the Pimote (ENER314) but I don't think that the gpiozero Energenie library supports the "two way" version (ENER314-RT), switching legacy devices with On Off Keying (OOK).

This repository https://github.com/whaleygeek/pyenergenie has support for my [legacy 4-gang switch (ENER010)](https://energenie4u.co.uk/catalogue/product/ENER010). This is the same 4gang switch as requested [here](https://github.com/theyosh/TerrariumPI/issues/199), but controlled by [this](https://energenie4u.co.uk/catalogue/product/ENER314) rather than [this](https://energenie4u.co.uk/catalogue/product/ENER314-RT)

This would be superb to support, as the ENER314 seems to be discontinued and the gpiozero library doesn't catch the replacement ENER314-RT

Thanks!

rossisbudda, 2021-07-01:
Also could be useful to know that the repository above got rolled into a node-red contribution. Not sure if that makes it any easier to pick up.

https://flows.nodered.org/node/node-red-contrib-energenie-ener314rt

Cheers :)

theyosh, 2021-07-19:
Hi,

Hmm, I am afraid that this will be difficult. As I do not have this kind of hardware, I cannot test it.

Also, on https://github.com/whaleygeek/pyenergenie it states that it needs to learn the commands from the RF signal. That means that you have to find the right values first, and maybe we can use that as the address.
And the software you are referring too, cannot be used in other software packages. As he did not make a package for it. Just some scripts.

So I am afraid that this is will be hard to make. For now, I cannot support it. I first want to finish the new version 4.

closed 2021-10-30 by theyosh