theyosh/TerrariumPI #560: [BUG]: databasa malformed after upgrade to 3.10

kaju666, 2021-07-06:
**Setup:**
 - Python version: 3
 - Raspberry PI: 4

**Describe the bug**
After running  sudo ./install.sh 3, terrarium don't wnat to run.
The error is database malformed. Can i somehow fix this or i need to reinstall everything from scratch?

**Screenshots**
[image]

theyosh, 2021-07-06:
Hi, this is fixable, but hard to explain.

But, if you can zip your DB and send it to me with wetransfer.com using email address github@theyosh.nl, then I will look into it. And I think I can restore it.

Else the solution is to delete (rename) the history.db file and restart. Then you are starting from scratch.

But the strange thing is, that this should not happen because of the update. I am afraid that this has happened due to the fact it has not shutdown correctly. Or, the error could be old, as the VACUUM command will check the hole database. And if the error is more then one month old, you would not notice it... as you are not requesting that old data on the graphs.

kaju666, 2021-07-07:
Hi

After Renaming history.db it started to work. Thanks.
And i think there is no need for you to repairing this DB, i even do not know what is in there :) I do not need any history data

closed 2021-07-07 by kaju666