theyosh/TerrariumPI #561: [BUG]: V4 BME280 not detected

chrRtg, 2021-07-08:
**Setup:**
 - Python version: 3
 - Raspberry PI: 3B

**Describe the bug**
A BME280 has not been detected by TerrariumPI. It showed up with `i2cdetect -y 1`
The problem was solved after I had installed the missing smbus with: `sudo apt-get install python3-smbus`

theyosh, 2021-07-09:
Hi,

thanks for testing. But this is a bit strange. As I use python virtual env and does not allow system wide packages. Therefore installing the python3-smbus with apt should not make a difference.. So I am a bit baffled about this.

As I install both smbus and smbus2 : https://github.com/theyosh/TerrariumPI/blob/4.x.y.z/requirements.txt#L21

Just to be sure, you have version 4 running?

chrRtg, 2021-07-09:
Yes, it's the 4.x.y.z with Rapsian lite, firmware upgrade and apt upgrade/update before install. 
to confirm: smbus and smbus2 are in ./venv/lib/python3.7/site-packages/

I would suggest do take this as a glich, close the issue as resolved.

theyosh, 2021-07-13:
Thanks for the feedback. Then the default solution is still valid: rerun the installer once more.

closed 2021-07-13 by theyosh