theyosh/TerrariumPI #562: [BUG]: V4 Database lock

chrRtg, 2021-07-08:
**Setup:**
 - Python version: 3
 - Raspberry PI: 3B

**Describe the bug**
TerrariumPI was not booting up. In the syslog I found the following lines:
`Jul  8 18:04:15 raspberrypi python[611]: sqlite3.IntegrityError: UNIQUE constraint failed: yoyo_lock.locked
Jul  8 18:04:15 raspberrypi python[611]: During handling of the above exception, another exception occurred:
Jul  8 18:04:15 raspberrypi python[611]: Traceback (most recent call last):
Jul  8 18:04:15 raspberrypi python[611]:   File "/home/TerrariumPI/terrariumPI.py", line 19, in <module>
Jul  8 18:04:15 raspberrypi python[611]:     terrariumEngine = terrariumEngine(__version__)
Jul  8 18:04:15 raspberrypi python[611]:   File "/home/TerrariumPI/terrariumEngine.py", line 88, in __init__
Jul  8 18:04:15 raspberrypi python[611]:     init_db(self.version)
Jul  8 18:04:15 raspberrypi python[611]:   File "/home/TerrariumPI/terrariumDatabase.py", line 26, in init
Jul  8 18:04:15 raspberrypi python[611]:     with backend.lock():
Jul  8 18:04:15 raspberrypi python[611]:   File "/usr/lib/python3.7/contextlib.py", line 112, in __enter__
Jul  8 18:04:15 raspberrypi python[611]:     return next(self.gen)
Jul  8 18:04:15 raspberrypi python[611]:   File "/home/TerrariumPI/venv/lib/python3.7/site-packages/yoyo/backends.py", line 324, in lock
Jul  8 18:04:15 raspberrypi python[611]:     self._insert_lock_row(pid, timeout)
Jul  8 18:04:15 raspberrypi python[611]:   File "/home/TerrariumPI/venv/lib/python3.7/site-packages/yoyo/backends.py", line 352, in _insert_lock_row
Jul  8 18:04:15 raspberrypi python[611]:     "(run yoyo break-lock to remove this lock)".format(row[0])
Jul  8 18:04:15 raspberrypi python[611]: yoyo.exceptions.LockTimeout: Process 805 has locked this database (run yoyo break-lock to remove this lock)
`
While I'm not used to yoyo or such I finally deleted the database file `terrariumpi.db`. After the next restart TerrariumPI came up properly.

I don't know if this is a bug or bad luck which broke the database. Maybe a short help on how to unlock or repair the database in such cases will help.

theyosh, 2021-07-09:
HI thanks for the issue,

I am a bit new to the yoyo package. It looked a nice DB migration package. So I have seen this issue myself, but I did not much investigating about this. As you also noticed, removing the terrariumpi.db and restart will fix it.

And when I remove my db and restart, it always works. So could it be that this has happened just right after the install? And where the installer is asking for a reboot?
Because I have 1 idea:
When you (re)start a Pi, I noticed that the clock is off. And it takes some time before the Pi syncs it time. And I have a watchdog for the terrariumpi service, which will restart if the startup does not take place in 2 minutes. So I do not know what happens when you boot your pi, which is 10 min behind, start my software, updates the clock and jumps 10 min. And maybe this will trigger the watchdog to restart my software. And that will make a second start, and then you get these DB locking errors.

Once you are running, and remove the db and start over, there is no time syncing issue, and no extra restart of my software... But this is a guess.. hard to test.

chrRtg, 2021-07-09:
Then it's an issue which may happen only on install because this yoyo is only used for a migration?

I understand TerraiumPI is a service. Then it should be stopped before a shutdown or reboot. Maybe the order is wrong, first stop watchdog, then TerrariumPI?

theyosh, 2021-07-13:
I have done some testing, and I can't reproduce. But for now, this only happens on an clean install. Then deleting the DB is the easiest thing to do. As there is no data in it, you delete the DB and let the software recreate.

I think this will be a FAQ entry.

theyosh, 2021-07-19:
Closing now. Will add it to the FAQ when final version is released.

closed 2021-07-19 by theyosh